LaserWeb/lw.comm-server #1: Comms diagram

cprezzi, 2016-12-19:
[image]

cprezzi, 2016-12-19:
The backend (lw.comm-server) will have one unified interface for the frontend and handle all different (machine) connection interfaces and firmware protocolls.

Machines can be connected through:
* serial (USB or UART)
* Ethernet (LAN / WLAN).

The following firmware will be supportet (at first):
* Grbl (min. v1.1e)
* Smoothieware
* TinyG

openhardwarecoza, 2016-12-19:
Slight correction: Ethernet on smoothie is a simple ajax/http_post
interface

Wifi is a Websocket server to serial running on an esp8266. (its also
firmware agnostic,  it just bridges serial to websocket and vice versa,
 and has a web interface to configure Wifi settings like AP and psk) so it
can even bridge Grbl

openhardwarecoza, 2016-12-19:
One way to unify wifi and ethernet: Smoothieware supports a telnet
interface...  Havent used it yet but apparently closer to the serial
interface than ajax (though ajax/post is the only way to upload gcode to
sd).
For the esp,  esplink firmware also has the same telnet interface that
bridges to serial

cprezzi, 2016-12-19:
Thanks for the hint. I'm still thinkering and writing down so I don't forget ;)

openhardwarecoza, 2016-12-19:
Same here,  dropping notes as i recall them (:

cprezzi, 2016-12-19:
TinyG is Websocket, if I remeber correctly.

cprezzi, 2016-12-19:
Good to know about esplink, thanks.

openhardwarecoza, 2016-12-19:
Tinyg is json over serial.  (for status and config)  - stringified json
anyway... But gcode gets fed same as others,  except instead of 'ok'  the
response is sr:*

cprezzi, 2016-12-19:
Yes, I know there is a serial communication for TinyG, but I thought there is also an ethernet connection. I will investigate :)

cprezzi, 2016-12-19:
Seems I'm wrong, only USB on TinyG and gShield (via Arduino Due).

cprezzi, 2017-02-15:
@openhardwarecoza Can you give me the communication diagram in original file format? I'd like to adjust it a bit and add it to the Wiki.

openhardwarecoza, 2017-02-18:
Lets see if the share works.  I drew it on Gliffy.com (free account) 
https://www.gliffy.com/go/publish/11458681

openhardwarecoza, 2017-02-18:
There is a Duplicate button on that window

cprezzi, 2017-02-19:
Jep, worked.
How did you export the graphic? Screenshot? I am not allowed to save as JPG (with free account).

closed 2017-02-28 by openhardwarecoza