LaserWeb/lw.comm-server #2: LW4 Integration

cprezzi, 2017-01-10:
I'm comming to the point where I need to integrate lw.comm-server into LW4 and need some help how to start. I still have no clue how the react/Redux stuff works. Especially where I have place my Code and how to access the gcode or save/lose Settings.

@openhardwarecoza, @tbfleming : can one of you give me a Quickstart?

tbfleming, 2017-01-10:
@cprezzi I just committed some example code to ```components/com.js``` to get you started on the UI part.

Note that to add new fields to settings, you need to add them to the bottom of ```reducers/settings.js``` or they'll get discarded on reload.

tbfleming, 2017-01-10:
@jorgerobles may also help answer questions.

cprezzi, 2017-01-10:
Thanks @tbfleming. That's a start.  
Now, as I understand the structure, the code to config the connection and initiate the connection has to go into com.js.  

Is it correct to place the code that handles the user interactions (like jog, play, pause...) in jog.js?

Is it ok to include a comm-server client library into the lib folder and use that lib from com.js and jog.js?

How do I access the gcode that has to be run?

tbfleming, 2017-01-10:
> Now, as I understand the structure, the code to config the connection and initiate the connection has to go into com.js.

yes

> Is it correct to place the code that handles the user interactions (like jog, play, pause...) in jog.js?

yes

> Is it ok to include a comm-server client library into the lib folder and use that lib from com.js and jog.js?

yes. lib/ is for anything that doesn't fit into the other folders. Other options include git submodules and npm packages.

> How do I access the gcode that has to be run?

I just pushed an example.

cprezzi, 2017-01-10:
Ok. Let's see how far I get ;)

jorgerobles, 2017-01-10:
_One piece a time, The Ripper said_

@cprezzi  think of `jog.js` as a widget. An autocontained piece of code that holds all the rendering and interaction. Interaction is on two folds, UI, and (application) State.
The component reads UI interactions (and optionally keeps his own state) and dispatches commands (Actions) to the Application state if needed.

So, to bind a library (as lw-comms) to a component is pretty straightforward.
If the client library is ready to be used, just jump into UI. Make it static first, and then add interaction.

If any setting should be remembered by the application, you have to bind it to the Application State thru `connect`  method wrapper. Many aspects of the state (in this case Settings) are developed already, so you only have to dispatch the correct action.

Anyway, ask any questions you have (on G+if needed), I'm here to help.

jorgerobles, 2017-01-20:
@cprezzi have you opened a branch on lw4 for the integration? Do you need help (react or other?)

openhardwarecoza, 2017-01-21:
I dont know if its any help,  but I'll post it nonetheless

From the days before @tbfleming joined the project, there was a 'dev' branch by @lautr3k which actually had a really nice comms module - maybe some work from there can make your tasks easier (probably not since its not react - well maybe some of the UI inspiration if nothing else)

[image]

Locked up in 
https://github.com/LaserWeb/LaserWeb4/tree/dev

openhardwarecoza, 2017-01-21:
(; i quite liked the icons in the console. He also had some websocket status info in the console, and automatic reconnect (see sept 8th)

https://github.com/LaserWeb/LaserWeb4/commits/dev

cprezzi, 2017-01-21:
I did not open a new branch in LW4 yet. I am working locally on a cloned dev-es6 and planned to push directly to dev-es6, as this is a development version and nobody else is working on the same components (com and jog). Is that ok, or should I move my work to a new branch?

@openhardwarecoza: Thanks for the hint. I didn't know sebastien allready worked on the frontend for comms. I will check it to see if it helps me. At least his styling is nice.

The basic communication (USB connect, send command to server, jog) is allready working. I am working on the details and stability now.

During development I realised that it is not very convenient to have comms in a separate tab of the sidebar. It was better in LW3, when it was always visible. Is it conceivable to have the "toolbar" also in LW4?

openhardwarecoza, 2017-01-21:
Top toolbar can be a component yes,  loaded from components/laserweb.js if
I recall correctly

tbfleming, 2017-01-21:
The CAM tab is already crammed for space; a toolbar will make this worse. Combining comms and jog sounds good to me.

cprezzi, 2017-01-21:
We could also move the config of the machine interface to the settings tab and only have the connect button and state in jog module. It's unlikely that users will change the machine connection very often.

jorgerobles, 2017-01-21:
@cprezzi Indeed, a tiny icon aside the connect with gears, by example could trigger the change to settings pane.

About the branching or not from dev-es6 was just to know your preference. Either if a branch is opened, or if you push to dev-es6, We could check / help with the integration if you need it. [You are not alone](https://www.youtube.com/watch?v=pAyKJAtDNCw)...

jorgerobles, 2017-01-25:
@cprezzi, @tbfleming I'm curious, what's is the way to bootstrap the server(ws)/client(app) with webpack?
Is it possible to serve WS and http on the same port?, If not, the WS server should be run on another port?
Will the server be launched (via npm script) and serve static` /dist`, or would it conflict with the hot-dev-server?

@cprezzi, I'm looking to the comm-server and maybe will be good to create an index with the verbs (i.e. describing the ws api) so we can collaborate implementing frontend/backend. What do you think? (grep `emit` will suffice as an alternative)

DarklyLabs, 2017-01-25:
@jorgerobles @tbfleming 
You might be interested in the method and results we obtained with raster engraving in this thread:
https://github.com/LaserWeb/LaserWeb3/issues/219#issuecomment-275093654

tbfleming, 2017-01-25:
@jorgerobles we'll probably keep the webpack dev server separate from the WS server since only devs need the dev server. They'll have to run on different ports if we do it this way.

cprezzi, 2017-01-26:
@jorgerobles I have made a wiki page on gh to document the server api (work in progress). Is that ok, or do you prefer another way.

I don't know much about webpack, so I leave that to the experienced people ;)
I have made the server IP:port configurable on client side, so it is possible to connect the same host:port as the webserver, or something different.

cprezzi, 2017-01-26:
@DarklyLabs For me the proirity is to bring the lw.comm-server to LW4 first, and then care about performance.

jorgerobles, 2017-01-26:
this one? https://github.com/LaserWeb/lw.comm-server/wiki/Client-communication
Fits perfect there :)

cprezzi, 2017-01-26:
Ok. I will extend it, once the api has sellted a bit ;)

cprezzi, 2017-01-31:
I could use some help. 

I need to fill the availabe serial ports to a dropdown by code. The actual form line looks like this:  
`<SelectField {...{ object: settings, field: 'connectPort', setAttrs: setSettingsAttrs, data: ['COM3',' COM4'], defaultValue: '', description: 'USB / Serial Port', selectProps: { clearable: false } }} />`  
How can I fill the ports by code instead of static data attribute?

jorgerobles, 2017-01-31:
Hi @cprezzi take a look to https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/webcam.js#L246

This is the most basic example of how a select is populated with programatic data.
Note that is not advisable to get data on render(). Render should use "static" data, through props or internal component state, else it could enter a loop because every change of data re-renders the component.
So basically you have to alter the internal state first (with setState()) and then the render makes its magic.
I found this package https://www.npmjs.com/package/react-websocket that encapsulate all the websocket boilerplate to a component. I think is a good idea-

cprezzi, 2017-01-31:
@jorgerobles The socket communication is not the problem. That's already working ;)

cprezzi, 2017-01-31:
I'm just fighting with some react/redux specific details.
I see that your sample is using pure select field instead of the SelectField of forms component. It seems that handling is different, right?

jorgerobles, 2017-02-01:
Yes, I had to use it because a weird glitch with other components in the pane. But anyways, the relevant code you should modify is the data attribute 

`<SelectField {...{ object: settings, field: 'connectPort', setAttrs: setSettingsAttrs, data: `**this.state.ports**`, defaultValue: '', description: 'USB / Serial Port', selectProps: { clearable: false } }} />`
being `this.state` populated with the websocket content.

If you are dispatching the action to setSettingsAttrs make sure `connectPort` is added to the settings reducer. Else won't pass state validation and will be discarded.

cprezzi, 2017-02-12:
How can I integrate the lw.comm-server (the server part) into the LaserWeb4 distro and also start the server in together with npm start?
At the moment I have to start the server in a separate command window.

jorgerobles, 2017-02-12:
Seems npm-run-all does this http://stackoverflow.com/questions/30950032/how-can-i-run-multiple-npm-scripts-in-parallel

cprezzi, 2017-02-12:
And to integrate lw.comm-server into LaserWeb4 distro? Is it enough to add lw.comm-server to devDependencies in package.json?

cprezzi, 2017-02-12:
How does npm know where to get it from?

jorgerobles, 2017-02-12:
@cprezzi really I dunno, I will dig some docs out there. :|

openhardwarecoza, 2017-02-12:
Hold a sec (: heading to my PC to give you the writeup.

(lw4 repo + comms repo,  into third repo (electron build env)  + some npm
scripts to a)  build /dist,  b)  build server.js into an electron app,  c)
 package as ASAR and d)  create installer

openhardwarecoza, 2017-02-12:
Ie the new repo with have lw4 and lw.comms as git sub's (not npm.  To
install a dependency using npm you have to publish it on npm.  We tried it
in lw3 for a while but npm and its node_modules path doesnt play nice with
electron-builder,  so we end up using git (subtree or submodules,  cannot
remember)  since that lends us control of the paths

openhardwarecoza, 2017-02-12:
Okay I am busy pushing the electron environment over to new repo https://github.com/LaserWeb/LaserWeb4-Installer

1) It has npm scripts that  ( `npm run installdev && npm run postinstall` )
1,1) puts laserweb4 into ./app
1.2) puts lw.comms-server into ./server
1.3) cd's into each and runs its npm installs + npm installdev + npm run bundle-dev to create /dist
1.4) copies that to ./release
1.5) starts electron `npm start` in ./release

2) once working locally you can also build an electron installer with `npm run dist`

Now this is just a rough copy of what we had in lw3 - now of course to fix it up to work with LW4

In server.js from LW3 we added 

```
// Electron app
const electron = require('electron');
// Module to control application life.
const electronApp = electron.app;

if (electronApp) {
    // Module to create native browser window.
    const BrowserWindow = electron.BrowserWindow;

    // Keep a global reference of the window object, if you don't, the window will
    // be closed automatically when the JavaScript object is garbage collected.
    var mainWindow;

    function createWindow() {
        // Create the browser window.
        mainWindow = new BrowserWindow({width: 1200, height: 900, fullscreen: false, center: true, resizable: true, title: "LaserWeb", frame: true, autoHideMenuBar: true, icon: '/public/favicon.png' });

        // and load the index.html of the app.
        mainWindow.loadURL('file://' + __dirname + '/public/index.html');

        // Emitted when the window is closed.
        mainWindow.on('closed', function () {
            // Dereference the window object, usually you would store windows
            // in an array if your app supports multi windows, this is the time
            // when you should delete the corresponding element.
            mainWindow = null;
        });
        mainWindow.once('ready-to-show', () => {
          mainWindow.show()
        })
        mainWindow.maximize()
    };

    electronApp.commandLine.appendSwitch("--ignore-gpu-blacklist");
    // This method will be called when Electron has finished
    // initialization and is ready to create browser windows.
    // Some APIs can only be used after this event occurs.

    electronApp.on('ready', createWindow);

    // Quit when all windows are closed.
    electronApp.on('window-all-closed', function () {
        // On OS X it is common for applications and their menu bar
        // to stay active until the user quits explicitly with Cmd + Q
        if (process.platform !== 'darwin') {
            electronApp.quit();
        }
    });

    electronApp.on('activate', function () {
        // On OS X it's common to re-create a window in the app when the
        // dock icon is clicked and there are no other windows open.
        if (mainWindow === null) {
            createWindow();
        }
    });
}

```

to the bottom of server.js ( might be cool to do an append after copy so the main repo for lw-comms isnt changed) https://www.npmjs.com/package/concat maybe)

I did a quick test and server isnt being copied 

I also recall there was some frontend fixes for LW3 (like https://github.com/LaserWeb/LaserWeb3/commit/093aa80c7b26fc31fd9121cef778f72b3766bb09) - so not there yet... but thats where we should be heading

openhardwarecoza, 2017-02-12:
Let me tag in the original masters who did the first round of work @krolco @copet80 (: if for nothing more than credit, but hey maybe we end up having questions

cprezzi, 2017-02-12:
I is already working :) 
I just had to change the following lines in package.json like so:
`"scripts": {
    "start-server": "node ../lw.comm-server/server.js",
    "start-app": "webpack-dev-server --progress --colors --open",
    "start": "npm-run-all -p -r start-app start-server",
...`

and:
`"devDependencies": {
...
    "lw.comm-server": "git+https://github.com/LaserWeb/lw.comm-server.git",
...
    "npm-run-all": "^4.0.1",
...`

openhardwarecoza, 2017-02-12:
> LaserWeb/lw.comm-server.git", ... "npm-run-all": "^4.0.1", ...

Ok,  I shall back off (:

cprezzi, 2017-02-12:
I have installed the npm-run-all package by `npm install npm-run-all` which automatically added the dependency to package.json. 
npm-run-all is used for starting multiple processes at the same time.

openhardwarecoza, 2017-02-12:
Go for it (:

cprezzi, 2017-02-12:
@peter: I'm not sure if I made it right, but at least it is working in the dev environment.

It would probably be good if the dev who configured the actual LW4 distro with webpack checks my changes. They are still only in my dev_comms branch at the moment.

openhardwarecoza, 2017-02-12:
Ahh, though rereading, seems you are still using webpack as web server...

Keep in mind the Electron thing is still a must so we can ship with installers. Users found installing nodejs + node-static + serialport hard enough, I dont want them to deal with Webpack as a webserver (:

Electron + comms server + static /dist built by webpack = installer

openhardwarecoza, 2017-02-12:
I'll keep working on the Installer repo for now - at least get it to build then we'll take it from there

cprezzi, 2017-02-12:
I only use the dev environment at the moment. I'm a noob with electron or what ever installation method we will use. 
I just thought that Sebasien or Todd did setup something else than Electron, because you had some problems earlyer, but don't remember exactly.

cprezzi, 2017-02-12:
By the way: You can test the comm implementation. Just use my branch `dev_comms`.
There are still some minor problems here and there, but basic functionality is working :)

cprezzi, 2017-02-12:
@openhardwarecoza : Be aware that the lw.comm-server package still includes an /app folder which is my LW3 like test environment. This folder is not needed in the installer.

tbfleming, 2017-02-12:
Right now there are three setups:
* npm start: devs only
* npm run bundle-dev: creates the standalone version with debugging info. Puts it in dist/. Requires a web server (not provided) or Electron. This is what's at https://laserweb.github.io/LaserWeb4/dist/
* npm run bundle-prod: creates the standalone version without debugging. Puts it in dist/. Currently broken; we'll probably distribute the debugging build for now.

tbfleming, 2017-02-12:
"standalone" = doesn't use webpack-dev-server

openhardwarecoza, 2017-02-12:
@tbfleming yip,  i noticed bundle-prod didnt work ( @lautr3k if you get a
chance? )

I used bundle-dev in my electron scripts.

DarklyLabs, 2017-02-13:
@cprezzi We just ran some tests with the dev_comms branch. Well done Claudio, fantastic to see LW4 controlling our machine.

cprezzi, 2017-02-13:
@DarklyLabs Did you also get the situation, that the serial connection was dropped when just left connected but idle for longer time? 
When that happens, I get a blue error messsage on the server and a windows sound of disconnecting & reconnecting USB.

DarklyLabs, 2017-02-13:
@cprezzi We didn't notice this situation because the machine was in use most of the time. We will test for this today. Roughly how long was it idle for when this happened?

We did notice that we needed to press 'abort' before generated gcode would actually run, even though the machine was idle and nothing was in the buffer. We will test this more today and advise if it is repeatable.

DarklyLabs, 2017-02-13:
@cprezzi The dev_comms version does not have all the latest features added by @jorgerobles.
Min/Max laser power for raster operations as well as his Camera tab are missing, for instance.
Which branch should we be using which has all most recent functionality?

cprezzi, 2017-02-13:
@DarklyLabs I did merge the newest changes in today. Just update. :)

DarklyLabs, 2017-02-13:
@cprezzi Into dev_comms?

cprezzi, 2017-02-13:
Yesterday I got the idle problem already after a few minutes, but today I tested with a different setup (smoothieboard and PC) and couldn't replicate it. So probably it just was the other setup (or settings).

cprezzi, 2017-02-13:
Yes, into dev_comms.

DarklyLabs, 2017-02-13:
@cprezzi Thx, just updated and it looks all there.

Two issues:
1: We need to press 'abort' before gcode will start running. This has to be redone every time. The console also reports runjob(undefined) whenever a job starts.

2: The console does not work. Commands appear to be sent in the console window, but machine does not react.

cprezzi, 2017-02-14:
@DarklyLabs I know there are still some state issues I have to fix.

1: This is caused by the alarm state that smoothie is reporting, but we don't have an alarm popup yet. I have implemented a automatic alarm clearing, but it doesn't work on all situations yet.

2: I'm aware of that. I'm still working on the console logs.

DarklyLabs, 2017-02-15:
@cprezzi is something broken today or is it something we are doing?

[image]

cprezzi, 2017-02-15:
It's still working on my PC. Try a `git pull` again and then a `npm run installdev`. Usually socket.io-client is always served by node.

[83 more replies]

closed 2017-02-28 by openhardwarecoza