LaserWeb/lw.comm-server #3: How to use get-bounds

cprezzi, 2017-02-16:
@tbfleming For the `check size` function in the jog tab I need the bounds of the actual gcode.
Can I use your get-bounds.js for this? If yes, how do I have to use it?

tbfleming, 2017-02-16:
get-bounds.js gets the bounds of a React component; no existing code gets gcode bounds. You may have to use tmpParseGcode.js.

cprezzi, 2017-02-17:
I realized that after writing the issue, sorry.
I just thought there must be some code, because you show the width and height of an object in the placing popup. Wouldn't it be best if the gcode class had these properties?

tbfleming, 2017-02-17:
The placing popup edits documents, not gcode. The Gcode class is dead code; it's the now-removed gcode tab.

cprezzi, 2017-03-03:
I made a simple solution by parsing all X and Y values in the gcode.  This is not perfect, because all moves are taken into account (incl. G0), which means that the bounding box includes the origin and not only the cutting part.

cojarbi, 2017-03-03:
I think that's good and similar to what we have in LW3. It useful in conjunction with a diode pointer or the test button to place the stock into location

cprezzi, 2017-03-03:
Right, that's why I implemented it and already pushed it to dev-es6 too ;)

closed 2017-03-20 by cprezzi