LaserWeb/lw.comm-server #5: Potential ID10T problem - "Machine connected" - even if port didnt respond with a valid machine string

openhardwarecoza, 2017-02-19:
Minor possibility, but stupid example (testing by thinking like a user)

1) If i connect to say com1 (onboard serial port -  nothing connected) - we actually report "Machine connected" - which is not true (:  .  Port was opened yes, but machine did not connect 

2) In lw3 I differentiated between "Port Openend" and "Port connected" - but that wasnt good enough either, the description was too similar to really indicate that there is a difference between a port opening, and a port actually connected (ie talking, getting replies back that we expect) 

[image]

cprezzi, 2017-02-20:
Probably I shouldn't call it "Machine connected" but rather "Serial connected" as this is, what it realy is.
A real sign of a working machine connection could be the firmware detection or just the ok after a valid command.

cprezzi, 2017-02-20:
Especially when connecting through ESP, the websocket connection to the ESP is used a the machine connection, but at this time we don't know if a machine is talking to the ESP.

cprezzi, 2017-03-03:
It's solved. If a new connection doesn't report a supported firmware, the connection is automatically closed after 2s.

closed 2017-03-03 by cprezzi