LaserWeb/lw.comm-server #6: Electron - Build

openhardwarecoza, 2017-02-21:
Morning

I integrated Electron-Builder into lw.comms-server itself 

1.  Basically, I copy laserweb4/dist to app/lw4 (we can later remove your test/app and replace it with lw4's html)

2.  I created a ./build/ directory to contain the icons/bg (hardcoded paths/filenames which electron expects to be what they are: http://electron.rocks/electron-builder-explained/)

3.  Then in server.js at the very bottom we handle the electron gui

4.  Lastly, in package.json I added two more scripts, and the build{} json for Electron-builder

I tested briefly and it didnt seem to break your normal workflows (lw4 start-all, and standalone lw.comm-server)

I also tested the built electron app and comms seem to be running.  

https://github.com/electron-userland/electron-builder/issues/974 used to be an issue, but as of this build its no longer an issue.  I suspect the bug got fixed somewhere in one of the dependencies and now just work (:  Yay

jorgerobles, 2017-02-21:
Great!! Hope to test today :)

openhardwarecoza, 2017-02-21:
Good stuff... Time crept up on us... @DarklyLabs starts shipping Emblasers this week, and @FabCreator will allow customers to collect their Fabkits on the 24th (shipping remainder the week after) - so I expect LW to have a rather busy week on the final testing of all this + support as users start receiving their machines

openhardwarecoza, 2017-02-21:
@Vertigotech - installer build instructions over on  https://github.com/LaserWeb/lw.comm-server/blob/master/README-ELECTRON.md - done - now you dont need to distribute rogue EXEs 
Please help test the build process and built app and once we are all happy (in a week or so) we can make EXEs/BINs/DMGs available for users to download directly from github.com/LaserWeb

openhardwarecoza, 2017-02-21:
Ahh, grr, found my first issue @tbfleming - dragging items on cam tab doesnt work.

openhardwarecoza, 2017-02-21:
Ok hold up (: I had Electron 1.4.1   - 1.6+ is on Chrome 56...

cprezzi, 2017-02-21:
@openhardwarecoza During testing I embedded lw.comm-server as dependency in LaserWeb4 and started both application toghtery (if needed) in packet.json of LaserWeb4. This way I was able to choose what I want to start with `npm start` (both) or `npm start-server` or `npm start-app`.

You flipped it around and included the app (LW4) in lw.comm-server (like it was on LW3).

But what about this scenario:
- Server started separately on a PC/Raspi... (not hosting the app)
- App installed (manual or elektron) on a PC/Notebook without server
- App connecting server on other machine. 

I would prefer to leave lw.comm-server as a separate, independent package (unified communications server) that can also be used by other frontends via the api.

Couldn't the electron compile also be made in LaserWeb4? This way we would have the options to compile installers with or without included server.

openhardwarecoza, 2017-02-21:
I tried that initially in LaserWeb4-Installer. Electron needs a node
backend to start,  and it needs a webserver running to host the UI on.

Feel free to refactor,  but nb we owe Darkly a working solution asap (last
week)  so quick wins to get going now is better than not going at all.

openhardwarecoza, 2017-02-21:
Honestly it hasnt made major changes except including lw's compiled dist in
the exe.  Apart from that change all i did was make node server.js into an
electron app (which would also be the best way to distribute lw.comm-server
as a standalone app (like spjs)

It already has a webserver.

In lw2 days i would have left out the webserver and told people to use
ghpages,  then run the server locally.  I also remember the backlash  that
decision had -  rioting users who dont have wifi in their garages...

The history of lw does make a lot of decisions for us

cprezzi, 2017-02-21:
At least you should make a separate electron branch and leave master slick and clean for separate usage.

openhardwarecoza, 2017-02-21:
Fine gimme 30 and i'll revert everything

openhardwarecoza, 2017-02-21:
Done

I'm out. Good luck

cprezzi, 2017-02-21:
@openhardwarecoza I didn't want to offend you in any way! I appreciate your opinion!!
I just wanted to stay slim with the server.

jorgerobles, 2017-02-22:
@openhardwarecoza, @cprezzi  I've got Peter's code from yesterday in my computer. I could open a branch. @openhardwarecoza do you agree? I will be sad if we lost the work you did. :cat:

openhardwarecoza, 2017-02-22:
Glad you did.  Git reset killed my local work

jorgerobles, 2017-02-22:
Available on https://github.com/LaserWeb/lw.comm-server/tree/electron_bundler

cprezzi, 2017-02-22:
@jorgerobles Thank you much!

@openhardwarecoza I am sad that you felt so negative about my statements. I did not mean it like that!

jorgerobles, 2017-02-22:
@openhardwarecoza could you restore the installation wiki page, plz? :smile:

openhardwarecoza, 2017-02-22:
Installation of electron?  Was a readme-electron.md inside the repo (: -  wasnt on the wiki

openhardwarecoza, 2017-02-22:
https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/README-ELECTRON.md

jorgerobles, 2017-02-22:
Oh, ok sorry, slipped that.

openhardwarecoza, 2017-02-22:
No problem.  I probably should have wiki'd it instead

jorgerobles, 2017-02-22:
I've added `git checkout electron_bundler &&\` because the branch is different now.
I've run the build script but get a blank screen (OSX) :|

openhardwarecoza, 2017-02-22:
Well, one of three things:

1) in https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/server.js#L1838 uncomment and see if the devtools give any clues

2)  Try electron 1.4x (I also noticed just before trashing my work https://github.com/LaserWeb/lw.comm-server/issues/6#issuecomment-281307654 - i was still using Electron 1.4x and that worked. When I uninstalled Electron 1.4 and installed 1.6, i also had a blank screen

2.1  If thats the case, then it means https://github.com/electron-userland/electron-builder/issues/974 might be back :(
2.1.1  running 'electron .' instead of npm run dist, should launch a live version:  If that works, but the built dmg/asar doesnt, then  its https://github.com/electron-userland/electron-builder/issues/974
2.1.2  If it also brings up a blank screen under live electron, its something else

3.  The server.js webserver on port 8000 serves the page that electron has to display:  So a quick check on chome :8000 should also help (if it displays in chrome, then server.js is serving it fine, but electron doesnt render it?

jorgerobles, 2017-02-22:
>    bundler/server.js#L1838 uncomment and see if the devtools give any
> 2.1 If thats the case, then it means electron-userland/electron-
> electron-userland/electron-builder#974
>    1. The server.js webserver on port 8000 serves the page that electron

Wow, thanks! I will report news ;)

cprezzi, 2017-02-22:
The lw.comm-server package is also included in lw4's package.json. Could that cause a problem when bundling it again with electron?

Also: LW4 uses port 8080 by default, but the webserver of lw.comm-server is on port 8000.

cprezzi, 2017-02-22:
On line 111 of server.js there is still the folder `./app` serverd as the web page. Shouldn't that be `./app/lw4` for the lw4 pages?
(see https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/server.js#L111)

openhardwarecoza, 2017-02-22:
@cprezzi nope thats still fine.   ./app is served by node static, but the webserver will handle subdirectories (otherwise /js and /lib wont get served either) 

As a test running node server.js in there, and accesing localhost:8000/lw4 should bring it up 

https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/server.js#L1825 loads up the window into electron

In this sense we really are running lightweight.  server.js still does everything it did (webserver, usb, etc) only, at the very bottom, we launch a electron window (aka fullscreen slimmed down chrome browser)

that loads up webserver content from server.js:8000

cprezzi, 2017-02-22:
@openhardwarecoza Ah ok, didn't see that electron specific line. Have you tried to use localhost instead of 127.0.0.1? I had that problem on my Win10 System.

openhardwarecoza, 2017-02-22:
I am busy re npm'ing it now (+-500mb total @60kbps link... driving me slowly insane)

cprezzi, 2017-02-22:
Ok, I leave you. I'm anyways a electron noob.

openhardwarecoza, 2017-02-22:
hehe no problem. we are only scratching the skin.... Once we get this ball through the gate theres still a hundred more (asar auto update servers with nutz/squirrel, being a high priority)

cprezzi, 2017-02-22:
I agree.

jorgerobles, 2017-02-22:
I think is 2.1.1 (974 issue) "./node_modules/.bin/electron ." works

openhardwarecoza, 2017-02-23:
@jorgerobles (: I'm still busy re-npm-ing (slow link = times out ECONNRESET while installing... grrr - even more reason to do the exercise of making it electron.  The webpack dev environment on LW4 and the electron build env total to over 600mb (CD full just to build the 37mb LW4 electron installer (;  

Was your test with electron 1.4.x as is currently in package.json, or 1.6.1 (latest) -  my working test the other day was under 1.4.6 but we cannot use it as chrome 54 (in electron .1.4.6) doesnt support the touchevents/dragevents used in LW4

openhardwarecoza, 2017-02-23:
Well, answering my own question, launching electron .  on 1.6.1. works (:

[image]

Lets see if a build works

openhardwarecoza, 2017-02-23:
Built and installed:  Working!

[image]

DarklyLabs, 2017-02-23:
@openhardwarecoza Can you post the exe for us to test?

openhardwarecoza, 2017-02-23:
@DarklyLabs https://drive.google.com/open?id=0B-UxBll2MapjTVNIVHlzcmdCa2s

openhardwarecoza, 2017-02-23:
(Win64 Build ... ) (We need to check if we can use Appveyor or something similar to automate builds for win32/64/linux/mac

DarklyLabs, 2017-02-23:
@openhardwarecoza Just installed it on Win10 64bit. Chrome Version 56.0.2924.87

Installed fine but just displays a white screen when run.

openhardwarecoza, 2017-02-23:
On first run, windows should bring up that Firewall allow access dialog.  Allow it, then restart laserweb

Also, while its running check in netstat if the webserver came up? netstat -an |find /i "listening" |find /i "8000"

[image]

DarklyLabs, 2017-02-23:
@openhardwarecoza Ok, that could be the problem. Firewall dialog is not appearing.

DarklyLabs, 2017-02-23:
[image]

DarklyLabs, 2017-02-23:
Any other dependancies I might need to have installed?

openhardwarecoza, 2017-02-23:
Check netstat 

https://github.com/LaserWeb/lw.comm-server/issues/6#issuecomment-281916238

openhardwarecoza, 2017-02-23:
Here's a build with the devtools turned on https://drive.google.com/open?id=0B-UxBll2MapjUnpvLVZmbFBIVkE - maybe there are clues

cojarbi, 2017-02-23:
Do we have a build procedure for OSx i can try?

cojarbi, 2017-02-23:
i get the sam as @DarklyLabs on Windows 10, firewall did appear and re checked is listening and granted access but the screen is white

cojarbi, 2017-02-23:
This is what i get with the dev tools
[image]

openhardwarecoza, 2017-02-23:
@cojarbi 

https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/README-ELECTRON.md should apply on Mac as well, except the very first step, instead of Windows Build Tools

Instead: 
- python (v2.7 recommended, v3.x.x is not supported) (already installed on Mac OS X)
- Xcode ( You also need to install the Command Line Tools via Xcode. You can find this under the menu Xcode -> Preferences -> Downloads This step will install gcc and the related toolchain containing make)

openhardwarecoza, 2017-02-23:
@cojarbi any clues on console tab?>

cojarbi, 2017-02-23:
@openhardwarecoza Console  tab is totally empty
[image]

openhardwarecoza, 2017-02-23:
[image]

Thats what I get.  One probable problem is maybe because LW4's Webpack bundle-prod isnt working and we use bundle-dev:  I see we are still launching some webpack dev server components. Maybe that causes issues

The socket.io errors is a little weird too as I can connect to the backend... but wont cause this symptom

openhardwarecoza, 2017-02-23:
Uploading a build with just the lw3 web interface - gimme 10 - slow internet

openhardwarecoza, 2017-02-23:
Ahaa, finally, I can replicate the issue (:

1) launching a new   mainWindow.loadURL('http://127.0.0.1:8000/anything');  from the exe, comes up blank
2) if I launch it first using 'electron . ', it works, subsequent runs with the exe also works (suspect we have loading issue from electron-builder still, running it once the electron . way caches the page... thats why it works.  

Great now that I can replicte, I can investigate

cojarbi, 2017-02-23:
I get this in the last step of the build npm run dist
[image]

openhardwarecoza, 2017-02-23:
try 'npm install -g electron-builder' then try again

cojarbi, 2017-02-23:
That did not run, it asked me to run it with admin, i did. Next i run npm run dist and get

Ariels-MBP:lw.comm-server ayahni$ npm run dist

> lw.comm-server@4.0.34 dist /Users/ayahni/lw4build/lw.comm-server
> npm run copy && build

> lw.comm-server@4.0.34 copy /Users/ayahni/lw4build/lw.comm-server
> ncp ../LaserWeb4/dist/ app/lw4/ --limit=6

Error: Config is invalid:
{
  "win": [
    {
      "remoteReleases": "Unknown option"
    },
    "Invalid option object"
  ]
}

Raw validation errors: [
  {
    "keyword": "additionalProperties",
    "dataPath": ".win",
    "schemaPath": "#/additionalProperties",
    "params": {
      "additionalProperty": "remoteReleases"
    },
    "message": "should NOT have additional properties"
  },
  {
    "keyword": "type",
    "dataPath": ".win",
    "schemaPath": "#/properties/win/anyOf/1/type",
    "params": {
      "type": "null"
    },
    "message": "should be null"
  },
  {
    "keyword": "anyOf",
    "dataPath": ".win",
    "schemaPath": "#/properties/win/anyOf",
    "params": {},
    "message": "should match some schema in anyOf"
  }
]
    at /usr/local/lib/node_modules/electron-builder/src/packager.ts:139:13
    at next (native)
    at /usr/local/lib/node_modules/electron-builder/node_modules/jsonfile/index.js:46:5
    at /usr/local/lib/node_modules/electron-builder/node_modules/graceful-fs/graceful-fs.js:78:16
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:445:3)
From previous event:
    at Packager.build (/usr/local/lib/node_modules/electron-builder/out/packager.js:278:11)
    at /usr/local/lib/node_modules/electron-builder/src/builder.ts:217:33
    at next (native)
From previous event:
    at build (/usr/local/lib/node_modules/electron-builder/out/builder.js:70:21)
    at Object.<anonymous> (/usr/local/lib/node_modules/electron-builder/out/cli/build-cli.js:71:41)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3

npm ERR! Darwin 16.4.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "run" "dist"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! lw.comm-server@4.0.34 dist: `npm run copy && build`
npm ERR! Exit status 255
npm ERR! 
npm ERR! Failed at the lw.comm-server@4.0.34 dist script 'npm run copy && build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the lw.comm-server package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     npm run copy && build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs lw.comm-server
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls lw.comm-server
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/ayahni/lw4build/lw.comm-server/npm-debug.log

openhardwarecoza, 2017-02-23:
Run git pull, then instead of npm run dist,   do npm run distmac (I added a --mac buildscript)

cojarbi, 2017-02-23:
getting this 
Ariels-MBP:lw.comm-server ayahni$ npm run distmac

> lw.comm-server@4.0.34 distmac /Users/ayahni/lw4build/lw.comm-server
> npm run copy && build --mac 

> lw.comm-server@4.0.34 copy /Users/ayahni/lw4build/lw.comm-server
> ncp ../LaserWeb4/dist/ app/lw4/ --limit=6

Error: Config is invalid:
{
  "win": [
    {
      "remoteReleases": "Unknown option"
    },
    "Invalid option object"
  ]
}

Raw validation errors: [
  {
    "keyword": "additionalProperties",
    "dataPath": ".win",
    "schemaPath": "#/additionalProperties",
    "params": {
      "additionalProperty": "remoteReleases"
    },
    "message": "should NOT have additional properties"
  },
  {
    "keyword": "type",
    "dataPath": ".win",
    "schemaPath": "#/properties/win/anyOf/1/type",
    "params": {
      "type": "null"
    },
    "message": "should be null"
  },
  {
    "keyword": "anyOf",
    "dataPath": ".win",
    "schemaPath": "#/properties/win/anyOf",
    "params": {},
    "message": "should match some schema in anyOf"
  }
]
    at /usr/local/lib/node_modules/electron-builder/src/packager.ts:139:13
    at next (native)
    at /usr/local/lib/node_modules/electron-builder/node_modules/jsonfile/index.js:46:5
    at /usr/local/lib/node_modules/electron-builder/node_modules/graceful-fs/graceful-fs.js:78:16
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:445:3)
From previous event:
    at Packager.build (/usr/local/lib/node_modules/electron-builder/out/packager.js:278:11)
    at /usr/local/lib/node_modules/electron-builder/src/builder.ts:217:33
    at next (native)
From previous event:
    at build (/usr/local/lib/node_modules/electron-builder/out/builder.js:70:21)
    at Object.<anonymous> (/usr/local/lib/node_modules/electron-builder/out/cli/build-cli.js:71:41)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3

npm ERR! Darwin 16.4.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "run" "distmac"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! lw.comm-server@4.0.34 distmac: `npm run copy && build --mac `
npm ERR! Exit status 255
npm ERR! 
npm ERR! Failed at the lw.comm-server@4.0.34 distmac script 'npm run copy && build --mac '.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the lw.comm-server package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     npm run copy && build --mac 
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs lw.comm-server
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls lw.comm-server
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/ayahni/lw4build/lw.comm-server/npm-debug.log
Ariels-MBP:lw.comm-server ayahni$

[40 more replies]

closed 2017-03-05 by openhardwarecoza