LaserWeb/lw.comm-server #9: Add option to change the server port

cojarbi, 2017-02-25:
To prevent conflict if the port is already in use by another server.
I had LW app running on one user and then loaded the app again from another user on the same machine issued the error but clicking on it did load the app. Maybe at this point or somewhere in the setting have tje option to change the port

cprezzi, 2017-02-25:
The error tells you only, that a second server tried to reserve the same port, which is not allowed by the os. On the app side, you can connect multiple instances to the same server. This is intended to connect from a mobile or tablet.

The server port is already configurable in config.js (in the lw.comm-server folder), but I'm not sure if this also works for the frontend.

cprezzi, 2017-02-25:
Would it make sense to prevent the start of a second instance of LW?

Another option could be to start the server only, if it is not already running.

cojarbi, 2017-02-25:
The feature to connect from multiple clients to the server is very important, but consider a user having another server running, this is not a major issue as i haven't seen it happen to many users in the past year but as LW gets an app and more features we could see this happening more often.

Both of your suggestions above make sense but think of making it possible to change the port maybe from the LW settings and write it back to the server. This would make it future proof

cprezzi, 2017-02-25:
I think changing the port from client sinde would not be a feasable solution, as it would need that the server and app already are working. Isn't it easyer to just change a value in the config file?

By the way, you can already change the server address:port to connect on from frontend. You just have to click on the "server connection" title to see the address field. This is made for connecting to a server on another machine (like a Raspberry Pi).

cprezzi, 2017-02-25:
@openhardwarecoza Is the config.js still usable (editable) with a electron installer?

cojarbi, 2017-02-25:
The functionality for connecting to the server from another client is amazing and its very clear ( at least to me) on how it works. With good documentation this should not be an issue.

Changing the port number on the config is also a solution ( on mac is not that apparent you need to dig dee into the app itself ) but its doable. 

Remember people dont read then again not sure how many times this could happen but it seems responsable to have a solution in place.

The app loads no matter what after showing that the address is in use so that is step one, could you the dialog the user to either know that theres another app running and change the port address ( send the change to the config file from there )?

Again this is not a big issue now im just looking ahead

openhardwarecoza, 2017-02-26:
Electron should still honor the environment variable (dotenv module is
still loaded?  Back from old work
https://github.com/LaserWeb/deprecated-LaserWeb1/commit/d6232d7eae02237953782dba20ee7fa559e0e7fb

cprezzi, 2017-02-28:
Ok, I have added env vars for the following config params:
WEB_PORT
DRO_DIGITS
VERBOSE_LEVEL (1-3)
LOG_LEVEL (0-3)

closed 2017-05-03 by cprezzi