LaserWeb/lw.comm-server #13: Inject into the serial stream

iceblu3710, 2017-03-02:
Is is possible to inject macros into the serial stream server side?

I will be running the comm server on a Raspberry Pi and would like to use the [OSMC remote ](https://osmc.tv/store/product/osmc-remote-control/) to jog around (using macros) and as the abort, feed hold, start buttons.

cprezzi, 2017-03-02:
It's already possible to have multiple clients connecting the server at the same time. Each client could send commands to the connected machine. So you could programm a small app that translates remote commands to gcode and sends them to the server via the websocket API.

We could also think about a plugin-structure for the server, but first we need to release LW4.

closed 2017-03-03 by cprezzi

cprezzi, 2017-04-11:
The newst version has keyboard jogging, so simulated keystrokes should also work.