LaserWeb/lw.comm-server #15: How to compile latest LW4 version in Electron

DarklyLabs, 2017-03-03:
@cprezzi @openhardwarecoza I have been able to successfully compile the windows exe following the process outlined, but the latest LW4 features from dev-es6 do not come through.

I also tried checking out dev-es6. as opposed to dev-comms and performing an electron compile, but this also does not bring across the latest LW4 fixes etc.

I notice we are up to 4042 for lw.comm-server but the electron_bundler branch shows 4036.

Any help to get this compiled would be much appreciated.

DarklyLabs, 2017-03-03:
If it helps:
1: running the dist with 'npm start' does run LW4 with the latest fixes.
2: the electron compiled version is 4.0.36

cojarbi, 2017-03-03:
Having exactly the same issues

cprezzi, 2017-03-03:
Peter has documented the process here: https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/README-ELECTRON.md

cprezzi, 2017-03-03:
But you have to replace dev_comms with dev-es6 to get the newst features.

DarklyLabs, 2017-03-03:
We are following Peters steps. 
We change it to dev-es6 and it does not implement them. 
See my original post.

cojarbi, 2017-03-03:
@DarkyLabs is this under windows 10. III m unable to build at all. I'm OSx I do get the build but ultimately 4036

DarklyLabs, 2017-03-03:
@cojarbi We can compile under Win 10 and OSX.
In both cases we only can get 4036.

DarklyLabs, 2017-03-03:
@openhardwarecoza Is there something else, other than ensuring dev-es6 is the checkout branch, we have to do to get the latest dev-es6 installers created?
We have users with machines and no software at the moment. Any help would me much appreciated.

openhardwarecoza, 2017-03-04:
Clear cache? (still looking for a programatic way but i open dev tools
inElrectron, and in the console tab, tick the bypass cache tickbox. Then
F5)
I had that before where i think i am not getting the nee code, but actually
its electron caching the old version

DarklyLabs, 2017-03-04:
@openhardwarecoza Sorry Peter, can you elaborate a little?
How do I open dev tools in Electron?

openhardwarecoza, 2017-03-04:
Settings tab, at the bottom
https://github.com/LaserWeb/LaserWeb4/issues/169#issuecomment-283069928
shows screenshot

openhardwarecoza, 2017-03-04:
> Settings tab, at the bottom https://github.com/
> LaserWeb/LaserWeb4/issues/169#issuecomment-283069928 shows screenshot

I'll try adding
https://github.com/electron/electron/blob/master/docs/api/chrome-command-line-switches.md#--disable-http-cache
today

DarklyLabs, 2017-03-04:
@openhardwarecoza Interesting situation. That option is not available on the electron build i can compile. So we need to clear the cache to have the latest option show up but the latest options need to be present to clear the cache. Inception?

The option are available when I run from 'dist'. The 'refresh window' works but the 'open dev tools' doesn't do anything.

cojarbi, 2017-03-04:
I do have those buttons inside settings / tools at the bottom but the dev
tools does nothing. Refresh does work

DarklyLabs, 2017-03-04:
@cojarbi Does reset reveal the latest features of LW?

cojarbi, 2017-03-04:
@DarklyLabs i dont se any

openhardwarecoza, 2017-03-05:
https://github.com/LaserWeb/lw.comm-server/commit/6579d84c3e9b156aacb77da6348ddab184d6b3ec#diff-78c12f5adc1848d13b1c6f07055d996eR1882  works on my test PC now. 

Will update readme in a minute to reflect dev-es6

closed 2017-03-05 by openhardwarecoza

cojarbi, 2017-03-05:
Ok it works now for me on OSX. It builds correctly to 4.0.42. 
Some notes if relevant
- The app start with developer tools open
- is this error from not having a connection?
[image]

openhardwarecoza, 2017-03-05:
Comment out devtools in server.js (near the bottom) once tested happy. 

That error is normal, is wds hotpack that doesn't have access to webpack dev server. Aka dev environment, can be ignored

DarklyLabs, 2017-03-05:
@openhardwarecoza  @cprezzi Completely deleted and re-created the build environment today for Windows build. Now getting the following error during the 'prepare electron modules'

× Rebuild Failed
An unhandled error occurred inside electron-rebuild
ENOENT: no such file or directory, open 'Z:\Users\DarklyBase\Sync\Projects\Emblaser2\Code\LaserWeb\lwbuildenv_win\lw.comm-serv
er\node_modules\serialport\package.json'

Error: ENOENT: no such file or directory, open 'Z:\Users\DarklyBase\Sync\Projects\Emblaser2\Code\LaserWeb\lwbuildenv_win\lw.co
mm-server\node_modules\serialport\package.json'
    at Error (native)

DarklyLabs, 2017-03-06:
@jorgerobles The settings tab doesn't open on either the win or mac electron build. Seems similar to what we have seen previously that you resolved.

cojarbi, 2017-03-06:
@DarklyLabs is something with the latest commit, i did not try to roll back but you could try and then build. If it does not work you can use this OSX i did before the commit

https://drive.google.com/file/d/0B-1--YftWEOrQUxUME9mVGxjOVk/view?usp=sharing 4.0.42

DarklyLabs, 2017-03-06:
@cojarbi Thanks. This version works correctly.