LaserWeb/lw.comm-server #16: IP Scan

DarklyLabs, 2017-03-10:
Similar to what is present in LW3.
Perform scan to find the IP address of all instances of a wifi-enabled smoothie/grbl device on the network.

We have some users with multiple machines and finding all instances of a smoothie firmware is important.

cprezzi, 2017-03-10:
WiFi / Telnet support (including IP scan) is comming with version 4.1. (See https://github.com/LaserWeb/LaserWeb4/issues/191)

DarklyLabs, 2017-03-10:
@cprezzi @openhardwarecoza Is there a simple program (win&mac) that users can run to determine their IP address? (Apart from checking on their router)
It would be good to have a way we can instruct non-technical users to find the correct IP for their machines.
Thanks for understanding.

cojarbi, 2017-03-10:
Why do you need to find the local IP address? Or you need to scan the network for the board?

To find local ip Maybe have them run local terminals its easier than having the download

Windows CMD - ipconfig
OSX Terminal - ifconfig |grep inet

To scan the network
OSX https://itunes.apple.com/us/app/lanscan/id472226235?mt=12

tbfleming, 2017-03-10:
```ipconfig``` isn't nice on new users. e.g. here's the output on my Surface Book:

```
Wireless LAN adapter Local Area Connection* 2:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Wireless LAN adapter Wi-Fi:

   Connection-specific DNS Suffix  . :
   IPv6 Address. . . . . . . . . . . : 2601:5c0:c100:4d50:203d:6c80:42d8:520d
   Temporary IPv6 Address. . . . . . : 2601:5c0:c100:4d50:d8f4:df94:5d9:2e17
   Link-local IPv6 Address . . . . . : fe80::203d:6c80:42d8:520d%8
   IPv4 Address. . . . . . . . . . . : 10.0.0.5
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : fe80::2e9e:5fff:fef4:1e95%8
                                       10.0.0.1

Ethernet adapter Bluetooth Network Connection:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Tunnel adapter isatap.{73580946-F435-413E-8A74-4C76E68C4602}:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Tunnel adapter Local Area Connection* 4:

   Connection-specific DNS Suffix  . :
   IPv6 Address. . . . . . . . . . . : 2001:0:9d38:6ab8:3cf0:2cb1:b627:1908
   Link-local IPv6 Address . . . . . : fe80::3cf0:2cb1:b627:1908%4
   Default Gateway . . . . . . . . . :
```

The output on my desktop is about 2-3 times longer.

cojarbi, 2017-03-10:
@tbfleming i know but many, including router manufacturers use them in their manual and describe to the user what are they looking for. Even 15 year old routers.

Of course as a user i prefer if you give me the easiest solution maybe in a form of a small app i can run

cojarbi, 2017-03-10:
any options here https://github.com/search?utf8=%E2%9C%93&q=ipscan&type=Repositories&ref=searchresults

DarklyLabs, 2017-03-10:
Sorry, please let me clarify. We need to find the IP address the laser machine is allocated when connected via WIFI.
This address is needed to connect to it via LW.

At the moment it is actually quite difficult to determine this address. 
Peter had implement a scan in LW3 that would find IP addresses that reported back information about the firmware I suspect.

Finding a device via the router is difficult. We run multiple routers here including apple products, netgear etc. Finding the correct IP of our machines via the router is hit and miss and not a good solution for regular users who are not technical.

cojarbi, 2017-03-10:
@DarklyLabs IPscan then but ( above a small git search for projects ) still it could an issue unless you tell them to look for the mac address whose first 3 pair are unique to a manufacturer

jorgerobles, 2017-03-11:
The ibox nano used wps to chain the lan and then shown it's ip on the lcd.
Just an idea...

cprezzi, 2017-03-11:
@jorgerobles I think @DarklyLabs has no display on the Emblaser2.

@DarklyLabs I suggest using AngryIP (from Ariel's link above) for the moment.

jorgerobles, 2017-03-11:
@darklylabs that could be a great addition (thinking on wemos esp8266 and
display under 3usd, or less)

domenic-d, 2018-01-22:
Closing this issue.

closed 2018-01-22 by domenic-d