LaserWeb/lw.comm-server #17: GRBL 1.1f Connection Issues

iceblu3710, 2017-03-10:
I am running the comms server from a Raspberry Pi and home made Arduino stepper shield. When LW connects to the serial port the Arduino does not restart and give the GRBL FW version which is how most program ID the firmware.

How does the comms server ID the firmware?

cprezzi, 2017-03-10:
lw.comm-server does also identify the firmware by the welcome string. Unfortunately we had to remove the automatic ctrl-x after connection because it causes smoothieware to go into alarm state. 
Just press CTRL-X after connecting. This should reset GRBL to get the welcome string.

cojarbi, 2017-03-10:
Could we add a field on the front end to be read by the server so the control-x or any other be executed ? 

⁣Ariel Yahni​

iceblu3710, 2017-03-10:
In the machine setup section we could have an "Reset On Connect" radio box or have it as a header and you choose which reset command you want to send.

For maximum flexibility we could have it as a textbox and let the user do whatever init stuff they wanted.

```
C^X ;Reset GRBL
M7 ;Turn on ventilation
```

cprezzi, 2017-03-10:
I worked very hard to get rid of all the firmware dependency in the frontend and I don't like to put it back now. 

This problem is only related to grbl directly connected by serial rx/tx (not USB). We don't have many users with such a configuration. 

I will find another elegant (automatic) solution. ;)

cprezzi, 2017-03-10:
This is related to issue #11

iceblu3710, 2017-03-10:
When monitoring the serial port on the pi with minicom and I connect to the device I get   

```
ok
error:2
error:2
```

When I inject `C^x` I get the reset string but the same error  
```
ok

Grbl 1.1f ['$' for help]
error:2
error:2
```

iceblu3710, 2017-03-11:
I struck oil!

So, I hijacked the comms-servers stdout messages to trigger a reset by doing this:

```
# npm run-script start-server -al | tee stdout.log
# tail -f stdout.log | grep -q 'INFO: Connecting to USB,/dev/ttyAMA0,115200' && echo -n $'\cx' > /dev/ttyAMA0
```

When I connect to the device it fires in the reset command before the comms server reads the first bytes.

A minor piping issue that may be an issue below.

```
pi@raspberrypi:~/Documents/LaserWeb4 $ npm run-script start-server | grep -q 'ttyAMA0' && echo $'\cx' > /dev/ttyAMA0
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: write EPIPE
    at exports._errnoException (util.js:1022:11)
    at WriteWrap.afterWrite (net.js:801:14)

npm ERR! Linux 4.4.50-v7+
npm ERR! argv "/home/pi/.nvm/versions/node/v6.10.0/bin/node" "/home/pi/.nvm/versions/node/v6.10.0/bin/npm" "run-script" "start-server"
npm ERR! node v6.10.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! laserweb@0.4.0 start-server: `node node_modules/lw.comm-server/server.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the laserweb@0.4.0 start-server script 'node node_modules/lw.comm-server/server.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the laserweb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node node_modules/lw.comm-server/server.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs laserweb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls laserweb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/Documents/LaserWeb4/npm-debug.log
```

iceblu3710, 2017-03-11:
I can flesh out the RasPi section of the docs as I work through my new machine setup. I wrote a script to start the server and setup the serial trigger so you can connect without caring.

```bash
#!/bin/bash
# Copyright (c) 2016 Trevor Johansen Aase
#
# The MIT License (MIT)
# Permission is hereby granted, free of charge, to any person obtaining a copy of
# this software and associated documentation files (the "Software"), to deal in
# the Software without restriction, including without limitation the rights to
# use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
# the Software, and to permit persons to whom the Software is furnished to do so,
# subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in all
# copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
# FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
# COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
# IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
# CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
#
# Useage:
#   ./start_server.sh /dev/YOUR-TTY

TTY=${1:-/dev/serial0}

function trigger {
  echo "Firing reset trigger to $TTY !"
  echo -n $'\cx' > $TTY 
}

echo "Starting LaserWeb4 Comms Server..."
npm run-script start-server | tee /tmp/lw.stdout.log &

while true
do
  tail -f /tmp/lw.stdout.log | grep -q --line-buffered "INFO: Connecting to USB,$TTY" && trigger
done
```

cprezzi, 2017-03-20:
Just for your info: The two errors you see are the trials to detect other firmware. You could just ignore them, as they should not harm your Grbl.

openhardwarecoza, 2017-03-21:
I gave @iceblu3710 edit permissions on cncpro.co  so I assume we can close this as a documented workaround for now.

Later we'll have an electron app with Pi support as a non-gui app

closed 2017-04-13 by cprezzi