LaserWeb/lw.comm-server #18: Enhancement:  Refresh Portlist

openhardwarecoza, 2017-03-21:
Motivated by https://plus.google.com/107113996668310536492/posts/4Y41g21bByK

In LW3 we had https://github.com/LaserWeb/LaserWeb3/blob/master/server.js#L506 and https://github.com/LaserWeb/LaserWeb3/blob/master/public/js/socketcomms.js#L120-L124 and https://github.com/LaserWeb/LaserWeb3/blob/master/public/index.html#L72

(:

But seeing as we are continually evolving (: - so I am wondering about future enhancement: 

https://www.npmjs.com/package/usb-detection + event to say "new device detected" 

Users often plug/unplug after starting the application

cprezzi, 2017-03-21:
I already have API commands for asking the server for available interfaces / ports with the corresponding events to the client, but I didn't implement the refresh button, because I was hoping to find a autodetection solution. Nice you found usb-detection! :) I will try it.

openhardwarecoza, 2017-03-21:
Super! Glad it helps

DarklyLabs, 2017-04-01:
@cprezzi cncjs has implemented auto-detection and connection which we found works well after being setup the first time.
https://github.com/cncjs/cncjs

Their streaming over USB works without hiccups to the smoothieboard, although no WiFi yet.

cprezzi, 2017-04-02:
@DarklyLabs Where do you get this info from? I didn't find any auto-detection code in cncjs and cheton marked it as 2.0 milestone (https://github.com/cncjs/cncjs/issues/127). The only interesting thing I found in cncjs is some sourcecode from me!

DarklyLabs, 2017-04-03:
Auto-connect is working.
Very useful.

cprezzi, 2017-04-03:
@DarklyLabs Auto-connect is not auto-detect, right?

DarklyLabs, 2017-04-03:
@cprezzi Auto connect allows the software to automatically connect to the machine when it is run without the user having to press 'connect'. It remembers the previous settings you have used.

cprezzi, 2017-04-04:
Hm. I don't like that feature. We have many Banggood/Eleksmaker diode laser engraver users and those machines switch the laser on for short time when the connection is opened. Therefore the user must consciously open the connection.
Additionally, we have users with multiple machines and they don't want to auto-connect to a machine, just because that was the last one they used.

cprezzi, 2017-04-21:
Port auto-refresh is implmented now. Connecting/disconnecting USB devices automatically refreshes the port list in LW4.

closed 2017-04-21 by cprezzi