LaserWeb/lw.comm-server #19: TravisCI Testing

openhardwarecoza, 2017-03-22:
@jonbev

Okay, so, the way we currently build LW4 binaries with Electron-builder is:

1.  This repo (lw.comm-server) has a branch, Electron_bundler, that has a couple npm script in package.json

See https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/package.json#L40-L51

The manual instructions are on https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/README-ELECTRON.md

(Might be worthwhile doing a manual build on your machine to familiarise yourself)

2.  The process also gets the /dist/ built fresh from the LaserWeb4 repo - the readme tells the whole story:
(pull LaserWeb4, run npm run bundle-dev to let Webpack build the /dist/, then copy that /dist/'s content to lw.comms-server(Electron_bundler branch)'s /app

jonbev, 2017-03-22:
Ok thanks, i'll give it a go.

jonbev, 2017-03-22:
OK so the manual build works but on running the AppImage file I get Lasweb running just get a white area where the grid normally is. Did I miss something?

[image]

openhardwarecoza, 2017-03-22:
Click Settings, Tools, Open devtools? Its supposed to work. If you get that
far, it's usually through the gate (:

Just ran the linux builds here (Ubuntu 16.04.2) and they work. (git upload
still running. Have you tested one of the existing appimages? If the viewer
doesnt show up it could be a webgl problem on the machine)

jonbev, 2017-03-22:
I just tried one of the builds you did and get the same so must be my machine.

I can't get to the settings tab either.

openhardwarecoza, 2017-03-22:
:( sorry

Just tested and can get to settings, viewer works, and devtools open from
the button under settings, tools. So must be.

probonopd, 2017-03-22:
Where can the AppImage be found?

openhardwarecoza, 2017-03-22:
In github.com/LaserWeb/LaserWeb4-Binaries?

Or on a local build? (lw.comm-server/dist/

probonopd, 2017-03-22:
Err, why are you putting binary downloads in git rather than on [GitHub Releases](https://github.com/LaserWeb/lw.comm-server/releases)? That's totally abusing git and GitHub certainly won't like it.

https://www.quora.com/Should-binary-files-be-stored-in-a-Git-repository

openhardwarecoza, 2017-03-22:
Thats what I said just a couple hours ago, which led to this new issue, to
start automating build and release. (; feel free to help out instead of
judging lol

openhardwarecoza, 2017-03-22:
Stalking you out on g+ you are actually exactly the skillset we have been
looking for https://plus.google.com/105493415534008524873/posts/HqrRabuEpjW

I'm fudging the builds in a a hacky way. We havent got a solution for
managing the versions of LaserWeb4 and lw.comm-server together into a
version to the name the builds (right now electron-builder gives them the
version from package.json in lw.comms-server

I'd be happy to give you push access if you can help us sort out the mess
for real (;

probonopd, 2017-03-22:
Happy to help if I can, but I am not an Electron expert at all.

openhardwarecoza, 2017-03-22:
The Electron part itself seems good. Automating building them (as per OP
above) and then sure, extending it to be able to automate releases would be
cool. We can maybe use git hooks on LaserWeb4 and lw.comm-server repos to
trigger the builds (if anything changes on there) or releases on there?
Thats the stuff we don't know what best practices!

@jorgerobles @cprezzi @tbfleming

openhardwarecoza, 2017-03-22:
Invited you to the Organisation, will full devteam access (:

probonopd, 2017-03-22:
So, would you like to have automated continuous builds on each `git push` on [Travis CI](https://travis-ci.org/)?

openhardwarecoza, 2017-03-22:
@probonopd - lets run that past @cprezzi  (developer of the comms-server) and @jorgerobles  / @tbfleming (the major frontend guys) for their call

openhardwarecoza, 2017-03-22:
Travis or something else?  Also not sure, we use node-serialport in the comms server and since its a native module, I couldnt get it to build crossplatform locally. I assume the same applies to the various CI services?

tbfleming, 2017-03-22:
It'd be nice if pushing to a specific branch caused it to create new binaries. It'd also be nice to have some way of marking known-good binaries.

jonbev, 2017-03-22:
I have it working when you push to the electron_bundler branch Travis will build the linux AppImage and upload it to an S3 bucket. I have also set a daily cron to do the same.

Here is the latest file that it built - http://laserweb-release.s3-website-us-east-1.amazonaws.com/laserweb-release/LaserWeb/lw.comm-server/10/10.1/dist/lw.comm-server-4.0.45-x86_64.AppImage

Clearly needs more work but at least there is something working.

jonbev, 2017-03-23:
I also have it building the AppImage for each commit to Laserweb4. If you click on the green tick next to the commit, it will take you to Travis, scroll down to the bottom of the log and there is a link to the artifact.

I'll see what I can do about cross platform builds next.

openhardwarecoza, 2017-03-23:
This is soooo awesome!

jonbev, 2017-03-23:
Just knocked up a lambda function to maintain a list of files in the bucket here -

http://laserweb-release.s3-website-us-east-1.amazonaws.com/

DarklyLabs, 2017-04-01:
@jonbev  Can you explain this thread (in non technical language) and how we can use it OR how it will fit into the big picture?

jonbev, 2017-04-01:
It basically just means that there is a build available for every commit. Just makes it easy for non technical people to use the latest build as they dont need to use git and npm etc.

DarklyLabs, 2017-04-01:
@jonbev Thanks for the explanation. Is it up and running yet? If so, where can we find the various builds?

jonbev, 2017-04-01:
I have only managed to get it working on linux build so far. I attempted windows but couldn't get it working as Travis runs on Ubuntu so made it very difficult. Could probably do with someone who has a clue about macs to have a look at the mac build as Travis does support them.

The link to a list of builds is here - http://laserweb-release.s3-website-us-east-1.amazonaws.com/ with the newest at the top.

DarklyLabs, 2017-04-01:
We are compiling mac builds locally using the procedure outlined by Peter. It works very reliably for Mac but we do have problems with Win builds and often have to re-clone and do some manual deletions to lw.comms directory to get it to pick up the latest versions and build correctly.

openhardwarecoza, 2017-04-12:
I updated the LaserWeb4-Binaries repo and removed the linux nighly builds. The readme on LaserWeb4 binaries now redirect to the S3 bucket

openhardwarecoza, 2017-04-24:
The linux builds are working so awesomely flawless? 

Any chance we can ask you to take another look at getting Windows/Mac builds working? (;

As I understand, TravisCI doesnt do Windows native modules, but AppVeyor can:  http://hoverbaum.gitlab.io/2016/08/03/Building-Electron-based-applications-using-CIs/ 

If its paywall related issues, let me know, and we can make a plan around those

openhardwarecoza, 2017-04-24:
PS: @probonopd - i also took your advice... Releases for Windows are now on https://github.com/LaserWeb/LaserWeb4-Binaries/releases

Linux builds are on the Amazon Bucket

Mac... Hmmm, well, for now, nowhere! Yikes!

probonopd, 2017-04-24:
Thank you very much. Unfortunately the nightly AppImage does not work for me:

```
me@host:~$ /home/me/Downloads/lw.comm-server-4.0.68-x86_64.AppImage 
2695
 
***************************************************************
        ---- LaserWeb Comm Server 4.0.68 ----        
***************************************************************
  Use  http://127.0.1.1:8000 to connect this server.
 
* Updates: 
  Remember to check the commit log on
  https://github.com/LaserWeb/lw.comm-server/commits/master
  regularly, to know about updates and fixes, and then when ready
  update accordingly by running git pull
 
* Support: 
  If you need help / support, come over to 
  https://plus.google.com/communities/115879488566665599508
***************************************************************
 
[3738:0424/201320.654755:FATAL:nss_util.cc(203)] nss_error=-5925, os_error=0
#0 0x000001cc9fbe <unknown>
#1 0x000001caf25b <unknown>
#2 0x000002fd78ce <unknown>
#3 0x0000007baf9b <unknown>
#4 0x00000072d2d7 <unknown>
#5 0x000000835bcb <unknown>
#6 0x000003c5ffc8 atom::AtomCertVerifier::AtomCertVerifier()
#7 0x000003c48da7 atom::AtomBrowserContext::CreateCertVerifier()
#8 0x000003ce2af4 brightray::URLRequestContextGetter::GetURLRequestContext()
#9 0x000002cd236c <unknown>
#10 0x000002deccd5 <unknown>
#11 0x000001cb26be <unknown>
#12 0x000001d04d0c <unknown>
#13 0x000001d04fb8 <unknown>
#14 0x000001d0531b <unknown>
#15 0x000001cb4a19 <unknown>
#16 0x000001d04aa7 <unknown>
#17 0x000001ce73f0 <unknown>
#18 0x000002de1de6 <unknown>
#19 0x000002de1f99 <unknown>
#20 0x000001d1cae5 <unknown>
#21 0x000001cf7543 <unknown>
#22 0x7f14098a46fa start_thread
#23 0x7f1402c54b5d clone
```

Possibly libnss3 and friends need to be bundled inside the AppImage then, also see https://github.com/probonopd/linuxdeployqt/issues/35.