LaserWeb/lw.comm-server #20: Error handling for EADDRINUSE

openhardwarecoza, 2017-03-23:
Quite regularly we get users who are trying to run two lw4 instances, or for example have lw3 open, then run lw4. 

CMD and Electron both popup a generic EADDRINUSE error.   Wondering if its possible for us to catch the error and rather print our own error message explaining to the user that an instance thats already open needs to be closed or something to that effect

openhardwarecoza, 2017-03-23:
http://stackoverflow.com/questions/9544479/how-do-i-catch-node-js-express-server-errors-like-eaddrinuse

openhardwarecoza, 2017-03-23:
https://www.nallex.com/questions/example-of-how-to-handle-eaddrinuse-errors-in-nodejs

openhardwarecoza, 2017-03-23:
[image]

DarklyLabs, 2017-04-03:
@cprezzi Hi Claudio, do you have any resolutions to capturing this error?
It is being reported by many LW4 users.

jorgerobles, 2017-04-03:
Maybe we can specify a range of ports and if catch error open other port (809x).
Downside could be the fill up of all the alternative ports. Have to say 8000/8080 are pretty common ports, maybe we can move to 90xx

jorgerobles, 2017-04-03:
Also maybe https://www.npmjs.com/package/node-netstat could be useful, scanning first if that port is used...

jorgerobles, 2017-04-03:
@DarklyLabs have you experienced that error? If so, could use a network diagnose tool to know what is on 8000? Maybe LW is not cleaning up correctly on binary or there's a common software that uses that port.

DarklyLabs, 2017-04-03:
Is it possible to only allow one instance of LW to run on a computer at any time?
We have found situations where the computer has gone to sleep and upon awakening chrome crashed but there is still an instance of LW running when we check the processes.

DarklyLabs, 2017-04-03:
@jorgerobles We can replicate the error by starting 2 instances of LW.

jorgerobles, 2017-04-03:
Yes, as it's using a specific port to launch, means that if is used conflicts each other. 
The problem here is the new instance cannot kill the former. Will investigate.

 Thanks for the feedback!

jorgerobles, 2017-04-03:
@cprezzi  https://github.com/electron/electron/blob/master/docs/api/app.md#appmakesingleinstancecallback

cprezzi, 2017-04-03:
@jorgerobles Thanks. I will insert the makeSingleInstance into the electron code. That's the most common reason for this error. I probably also need to implement a netstat check if the port is already used by another server.

cprezzi, 2017-04-03:
I have implemented the makeSingleInstance function to prohibit starting multiple instances. Instead, the main window is activated.
See commit https://github.com/LaserWeb/lw.comm-server/commit/905cde1913641ec99b8075baaa1af299c9c7792a

closed 2017-04-09 by cprezzi