LaserWeb/lw.comm-server #22: Enhance the "No supported firmware found" message

openhardwarecoza, 2017-04-07:
If it detects Grbl, but the version is too old, please let it log the error accordingly, something like "You are running Grbl 0.9 (Show actual version string) :  This application needs at least Grbl 1.1. Please update the version of Grbl on your controller, then try again"  or something like that? Right now if you connect to Grbl0.9 it just closes the port (so you have to use putty or something like that to check what version is actually installed, after taking a minute to realise thats why it closed it)

cprezzi, 2017-04-07:
You should get the error message `Grbl version too old -> YOU MUST INSTALL AT LEAST GRBL 1.1e` in the log area. Isn't that enough?

openhardwarecoza, 2017-04-07:
Hmm. Just now when I tested with Grbl 0.9 (forgot controller hasnt been
upgraded) it gave a No supported firmware message. But will retest once
rewired. Stand by (:

closed 2017-06-01 by openhardwarecoza