LaserWeb/lw.comm-server #30: Smoothieware List SD and Play SD functionality

openhardwarecoza, 2017-06-01:
websocket -> listSD -> runs a http://smoothieware.org/console-commands#ls, (try and get results recursively? even if it means manually going deeper) -> Returns ws event to frontend with array of files.

Frontends displays list of files with a PLAY button
Play button sends ws event to lw.comm-server: to http://smoothieware.org/console-commands#play

cprezzi, 2017-06-04:
If we do that, we should also make file upload available, even if it is dead slow.

openhardwarecoza, 2017-06-04:
Agreed.

openhardwarecoza, 2017-06-04:
See
https://github.com/LaserWeb/deprecated-LaserWeb3/issues/138#issuecomment-253009463
- in testing it was faster than M28 indeed. If you can, please implement it
so it can be used with esp and ethernet too?

cprezzi, 2017-06-06:
I have started implementing SD support for Smoothieware and MarlinKimbra in branch SD_support :)

openhardwarecoza, 2017-07-03:
Couple checklists to SD implementation 

[ ] track progress: https://github.com/openhardwarecoza/CoPrint/blob/gh-pages/public/js/socketcomms.js#L152 (when running sd, same as playing, disable jog, show progress (instead of queue) etc)

[ ] reenable jog etc once done running sd job

[] custom icons for gcode, config, firmware (just makes it more user friendly 

[ ] upload

[ ] list sd files

[ ] select a file to play it

[ ] ability to abort /pause job while running off sd (existing buttons)

openhardwarecoza, 2017-07-03:
Bonus points. Can we also make it work over the Telnet implementation? I am still working on the esp8266 bridge (boards came in last week) (:

openhardwarecoza, 2017-07-03:
Cool, i see @cprezzi already has pause, stop, progress, even rm and mv https://github.com/LaserWeb/lw.comm-server/blob/SD_support/server.js#L2157 in backend. So the above notes was more useful for @jorgerobles (; for frontend

jorgerobles, 2017-07-03:
Sorry I was away. Gonna review all the thread

openhardwarecoza, 2017-07-03:
Hehe yeah sorry, some of it happened in #30 too

jorgerobles, 2017-07-03:
[image]

jorgerobles, 2017-07-03:
Basic stub available on https://github.com/LaserWeb/LaserWeb4/tree/filemanager
@cprezzi

jorgerobles, 2017-07-03:
@cprezzi if can prepare the file tree as https://github.com/woodpig07/react-nested-file-tree will be fine
Don't worry for nesting. you could do only current directory and will sort out working.
@openhardwarecoza I will try to resemble coprint, as this component is very flexible.

openhardwarecoza, 2017-07-03:
No problem (: was just mentioning it as a sort of 'did work' UI. (;

Tree looks nice, might just hurt performance since you'd need to get list
of current dirz then seperately ls each subdir

cprezzi, 2017-07-03:
I think Peter is right about performance. It could be very slow to recursively scan the whole SD card to build the tree object for the filemanager.

jorgerobles, 2017-07-04:
Hmm the component only paint the data sent. Send one folder content and
that's all.

cprezzi, 2017-07-04:
@jorgerobles Does it make sense to use this lib, if we don't need the tree feature? I don't like unnecessary dependencies (Overhead). We just need a list with buttons.
I think I'll try to implements Peters list in the modal dialog ;)

jorgerobles, 2017-07-04:
@cprezzi this lib does not force be recursive. It's easy to show a single folder contents without recursion, and `folderClickHandler` could do whatever is needed,

-  as loading the inner folder showing recursion; or
-  wipe painted data and show the inner folder.

In either way, performance will not be a problem. The first way should update the state contents of the opened branch upon loading. The second way will replace all the state data, bringing the same "one folder" appearance.

jorgerobles, 2017-07-04:
Anyway.. is your call 😄

cprezzi, 2017-07-04:
I was concerned about the performance of the serial communication, not the frontend component.
We'll see what I can make out of it ;)