LaserWeb/lw.comm-server #31: Detect if its a laser or a mill?

openhardwarecoza, 2017-06-01:
Run http://smoothieware.org/console-commands#config-get for the laser_module parameter (if true, its a laser) also could probably check for extruders if found, is a 3d Printer.  Check for a Spindle config, its a mill... etc

On Grbl we can check for Laser Mode (if found its a laser. If not its a mill)

Then in the frontend (@jorgerobles) we can hide laser ops and "laser test" etc and any other instance of "laser" if its not a laser

tbfleming, 2017-06-01:
This sounds good for limiting what's on the Control tab. I'm uneasy with it limiting things on the Files tab.

openhardwarecoza, 2017-06-01:
If i may ask? Why? We have a cnc mode switch that hides cnc mill ops (;

openhardwarecoza, 2017-06-01:
Sidenote: it could be true on all 3 of course too

tbfleming, 2017-06-01:
I hate that switch. It contributes to people coming from jscut thinking it's not a complete replacement. They try it, see only laser ops, then quickly give up.

openhardwarecoza, 2017-06-01:
Yip me too, sort of why I want the switch replaced with "what is this machine based on what the firmware tells us"?

openhardwarecoza, 2017-06-01:
Also, want to remove the phrase Laser wherever not absolutely needed - since it scares of mills users seeing a Laser test button, laserweb title bar, etc

jorgerobles, 2017-06-01:
I will ditch da switch. Was only because LW3 had.

tbfleming, 2017-06-01:
I see this happening:
* Beginner tries CAM first
* Connects to a board that they haven't finished configuring
* Gets totally confused that CAM suddenly stops working right

openhardwarecoza, 2017-06-01:
Agreed, quick win there.  In default mode we can show all machine ops.  After connecting, we know what we are talking to, hide irrelevent ops?

openhardwarecoza, 2017-06-01:
Yeah ok, i hear you (;

openhardwarecoza, 2017-06-01:
I'll leave my root problem - just in case it inspires a different elegant solution to the problem:

I have several machines in my shop:  would be cool if LW adapts to the particular machine I connected to, when I connect to it.  On the one hand I thought, simple enough, check the firmware to see what tools are configured.   The other idea I had was to find a hash / serial number / something from the serial port, and automatically switch Machine Profiles based on the ID of the port you connect to (and no port name is not good enough, /dev/ttyUSB0 isnt always the mill (; )

jorgerobles, 2017-06-01:
Understood. I think machine should connect on user interaction, once connected, there's a work in progress https://github.com/LaserWeb/lw.comm-server/wiki/Firmware-feature-list that might solve that.

cojarbi, 2017-06-01:
I have the same issue as Peter, but what happens if I want to work in another or not connected? Or shows all ops on now connection, swtich switch to specific when connected

jorgerobles, 2017-06-01:
I'd go for change upon user actively chooses the machine to connect.

cprezzi, 2017-06-01:
I belief that the user should always initiate the connection because on some machines it's not save to connect without user knowing it (i.e. eleksmaker machine activates laser on connect!).

cprezzi, 2017-06-01:
The planned getFeatureList command could be easily extended by featuresLaser, featuresMilling or a "serial number" if I can detect those infos somehow.

What about having a firmware configuration dialog in LW4?

cprezzi, 2017-06-01:
If we can detect some machine serial number, we could name that machine and show that name in the COM-Port dropdown.

cprezzi, 2017-06-01:
The problem with reading the config (like laser module, spindle module, laser mode...) from the firmware is, that we don't know if the user has configured the firmware correctly.

openhardwarecoza, 2017-06-01:
@cprezzi
https://github.com/EmergingTechnologyAdvisors/node-serialport#serial-port-list
shows Serial number as a result.

Sidenote, all Smoothieware boards have the same pid and vid. Seeing those
immediately allows you to assume that port is a smoothie (;

cprezzi, 2017-06-18:
What do you think about that:

- If the user would have to define a machine profile for every machine, not one profile per machine type, we could store the last machine settings (read from firmware) in the machine profile.
- On every connection, we could compare the machine settings with all profiles and select the first maching one.

This way, we could store all "flags" in the machine profiles.

cprezzi, 2017-06-18:
And the other way around: 

- If the user selects a machine profile, we could try to open the last known port for this machine.

jorgerobles, 2017-06-18:
I'm not sure. I think machine hardware should have a command to identify
their purpose, defined by user. Not so hard ... :)

cprezzi, 2017-06-18:
On grbl, the user can store an idetification string with $I=xxxxx, but Smoothieware and TinyG don't have such a feature, as far as I know.

cprezzi, 2017-06-18:
What I mean is, that the frontend anyways has to save the settings for his features, so why not in the machine profiles. If there is a machine idetification number (or "serial number") we could store it in the corresponding machine profile, so we know which settings to load.

jorgerobles, 2017-06-18:
I agree. Some vendor machines will come with the features preloaded on
their profiles, some other belongs to the firm capabilities. Machine
profiles are meant for that, but we have to ask the machines - who are you?
- hope we can get help for other firms than grbl for that.