LaserWeb/lw.comm-server #33: Trying to Unify electron_bundler branch and master

jorgerobles, 2017-06-09:
Hi. I've observed some growing differences on electron_bundler and master. I've patched electron_bundler but if you could, please take a look to try to ammend branches, or we have to find a new configuration patter to binaries.

cprezzi, 2017-06-09:
What growing differences do you mean (beside of the additions I made)?

jorgerobles, 2017-06-09:
Dissapeared some bundle scripts,  app info and changed conflicting libraries on merge :/

cprezzi, 2017-06-09:
So mainly changes in package.json?

cprezzi, 2017-06-09:
How could we tell git not to merge package.json?

tbfleming, 2017-06-09:
I don't think git has an option like that. Plus, we do add dependencies to package.json which should be merged.

jorgerobles, 2017-06-09:
Maybe we should approach a bootstrap that uses server packaging decoupling
electron build from server from frontend. Don't know if makes sense, thought

cprezzi, 2017-06-09:
@tbfleming I was talking about package.json of lw.comm-server, which doesn't have many dependencies.

cprezzi, 2017-06-09:
@jorgerobles What does electron exactly need to host the frontend and start the node server? Could that also be bundled from LW4 repo? And would that make more sense?

cprezzi, 2017-06-09:
@jorgerobles I have now merged the contents of package.json from electron_bundler back to master.
To avoid further problems, we should make all changes on master and merge them later to electron_bundler.

openhardwarecoza, 2017-06-12:
How about packing lw.comm-server as electron too, but listen for say --no-gui switch then just dont run the Electron.Window code?

openhardwarecoza, 2017-06-12:
ie, wrap https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/server.js#L2670-L2741 in a IF

openhardwarecoza, 2017-06-12:
In terms of package.json, electron build only has more dependencies and the build scripts.  Apart from a larger initial npm install, may as well be the same package.json for Master.   If you consider only distributing comms-server as per https://github.com/LaserWeb/lw.comm-server/issues/33#issuecomment-307820923 then no one needs to npm install anyway except the devs who build the bins

jorgerobles, 2017-06-12:
Maybe we are talking the same and we don't know. Compiling Lw-comm-server as electron will weight a lot. @cprezzi  https://github.com/LaserWeb/lw.comm-server/issues/33#issuecomment-307447289 comment could work, trying again to translate electron build to Frontend, keeping Server as a package. On the other side, frontend has no binary dependencies yet.

Or maybe we are just looking for green dogs. Having the package.json solved could be just fine.

cprezzi, 2017-06-12:
@openhardwarecoza The Electron code is not needed in the master branch, as changes to server.js will never overwrite the electron code in the elevtron_bundler branch when merged.

cprezzi, 2017-06-12:
We have some users that use lw.comm-server standalone, on raspi for exampe, and I don't like to blow up the package size (and potential errors) for those users.

cprezzi, 2017-06-12:
Package.json in master is now near identical as in electron_bundler.
see https://github.com/LaserWeb/lw.comm-server/commit/70ae8e1b8d7d942cc26766b98c366678aef89229

closed 2017-06-18 by cprezzi