LaserWeb/lw.comm-server #35: Errors building LW

domenic-d, 2017-06-13:
@cprezzi @jorgerobles @openhardwarecoza 
I am unable to build LW today following the normal process.
It appears there is an error that now occurs during the lw.com-server build at the command:
./node_modules/.bin/electron-rebuild

I created fresh clones and ran the complete build process.

Can anyone verify the builds are currently working and if so, what could be causing this for me. I am building on mac.

cprezzi, 2017-06-13:
I have merged the last changes from master to electron-bundler (needed to cleaned some merge conflicts) and was able to build the installer.

Could you pull and try again please?

domenic-d, 2017-06-13:
@cprezzi I just created a fresh install and ran the build script.

Got the following error:
./first-build.sh: line 13: ./node_modules/.bin/electron-rebuild: No such file or directory

My first-build script is the following:
```
git clone https://github.com/LaserWeb/lw.comm-server.git &&\
git clone https://github.com/LaserWeb/LaserWeb4.git &&\
cd LaserWeb4 &&\
git fetch && git checkout dev-es6 &&\
npm install --verbose &&\
npm run installdev &&\
cd .. && cd lw.comm-server &&\
git fetch && git checkout electron_bundler &&\
npm install --verbose &&\
cd .. && cd LaserWeb4 &&\
npm run bundle-dev &&\
cd .. && cd lw.comm-server &&\
./node_modules/.bin/electron-rebuild &&\
npm run dist
```

domenic-d, 2017-06-14:
@cprezzi @jorgerobles 
The latest binaries (v4.0.734-96) appear to have a problem as well. They are listing as lw.comms-server as what is installed.

When running lw-comms-server, it appears LW runs.

cprezzi, 2017-06-14:
@domenic-d The error message sais that electron-rebuild is missing, which is part of electron.
Try to reinstall electron with 'npm install -g electron' (probably you need sudo...)

cprezzi, 2017-06-14:
Sometimes, i also need to run 'npm update' in LaserWeb4 folder to get all updates.

cprezzi, 2017-06-15:
@domenic-d Did you get it to work?

domenic-d, 2017-06-15:
@cprezzi 
No. Unfortunately not.
I am at a loss. Just tried building again on Windows and it fails there too. Neither mac or win machine has changed.
Has anything changed recently with respect to building?
Are there any other requirements or repositories that need to be present?

cprezzi, 2017-06-15:
@domenic-d We had an accidental overwrite of package.json in the electron_bundler branch, which was first restored by @jorgerobles and then I merged most of this file to the master branch, so it should not cause such a problem again.

There are a lot of precautions to be able to compile LaserWeb4 (all the dependencies) but lw.comm-server is relatively independant. 

Are you able to do all the stuff in LW4 folder, like 'npm install', 'npm run installdev' and 'npm run bundle-dev' and does LW4 work when 'npm start'?

openhardwarecoza, 2017-06-15:
Since the error relates to Electron-rebuild, its more likely npm install's
on the lw.comm-server than LW4 repo:  Run through
https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/README-ELECTRON.md
again and make sure its all installed, particularly, inside the
lw.comm-server branch, git checkout electron_bundler && npm install
--verbose

domenic-d, 2017-06-15:
Did a clean install again.
**Prepare Temporary Build environmen**t commands ran correctly
**Prepare LW4 /dist** commands ran correctly
**Prepare Electron modules** failed
-bash: ./node_modules/.bin/electron-rebuild: No such file or directory

cojarbi, 2017-06-15:
@domenic-d I also don't get that command to run on windows but proceed with npm run build and it builds correctly

domenic-d, 2017-06-15:
When I try bypassing that command I get the following:

```
> lw.comm-server@4.0.96 dist /Users/DarklyBase/Projects/mybuild/lw.comm-server
> npm run copy && build

> lw.comm-server@4.0.96 copy /Users/DarklyBase/Projects/mybuild/lw.comm-server
> ncp ../LaserWeb4/dist/ app/ --limit=6

sh: ncp: command not found
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! lw.comm-server@4.0.96 copy: `ncp ../LaserWeb4/dist/ app/ --limit=6`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the lw.comm-server@4.0.96 copy script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/DarklyBase/.npm/_logs/2017-06-15T22_28_17_605Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! lw.comm-server@4.0.96 dist: `npm run copy && build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the lw.comm-server@4.0.96 dist script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/DarklyBase/.npm/_logs/2017-06-15T22_28_17_633Z-debug.log
```

tbfleming, 2017-06-15:
ditto. I suspect electron reworked their build system on the latest release.

tbfleming, 2017-06-15:
This worked for me (Windows 10):

```
npm install electron electron-builder electron-rebuild ncp
node_modules\.bin\electron-rebuild
npm run copy
node_modules\.bin\build
```

domenic-d, 2017-06-16:
@tbfleming I can confirm your sequence builds correctly on my Win 10 machine as well.

On OSX it works up to the very last line where the following error is reported:

```
DarklyBaseOSX:lw.comm-server DarklyBase$ node_modules/.bin/build
electron-builder 18.6.2
Error: Package "electron" is only allowed in "devDependencies". Please remove it from the "dependencies" section in your package.json.
Package "electron-builder" is only allowed in "devDependencies". Please remove it from the "dependencies" section in your package.json.
Package "electron-rebuild" is only allowed in "devDependencies". Please remove it from the "dependencies" section in your package.json.
    at Packager.checkMetadata (/Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/src/packager.ts:381:13)
    at /Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/src/packager.ts:169:10
    at Generator.next (<anonymous>)
    at runCallback (timers.js:800:20)
    at tryOnImmediate (timers.js:762:5)
    at processImmediate [as _immediateCallback] (timers.js:733:5)
From previous event:
    at Packager.build (/Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/out/packager.js:284:11)
    at /Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/src/builder.ts:198:33
    at Generator.next (<anonymous>)
From previous event:
    at build (/Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/out/builder.js:70:21)
    at Object.<anonymous> (/Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/out/cli/build-cli.js:71:41)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Function.Module.runMain (module.js:605:10)
    at startup (bootstrap_node.js:158:16)
    at bootstrap_node.js:575:3
```

domenic-d, 2017-06-16:
@cprezzi To answer your earlier question. No problems are reported in the Laserweb part of the process.

cprezzi, 2017-06-18:
I alos assume that the problem is caused by electron, as @tbfleming said.

@domenic-d From your second last log it seems that the ncp command is missing. Which shell are you using? On my system it is different if I'm using git bash or a windows shell. On one of them, ncp is working, on the other I have to use cp.

cprezzi, 2017-06-18:
@domenic-d On the Mac you would have to use `node_modules\.bin\build --mac` (or use `npm run distmac`).

domenic-d, 2017-06-18:
@cprezzi I am compiling from 'Terminal' on the mac as has always worked in the past.
Neither of those suggested commands work.

the first returns this error:
`-bash: node_modules.binbuild: command not found
`
The second returns:
```
> lw.comm-server@4.0.96 distmac /Users/DarklyBase/Projects/mybuild/lw.comm-server
> npm run copy && build --mac 

> lw.comm-server@4.0.96 copy /Users/DarklyBase/Projects/mybuild/lw.comm-server
> ncp ../LaserWeb4/dist/ app/ --limit=6

electron-builder 18.6.2
Error: Package "electron" is only allowed in "devDependencies". Please remove it from the "dependencies" section in your package.json.
Package "electron-builder" is only allowed in "devDependencies". Please remove it from the "dependencies" section in your package.json.
Package "electron-rebuild" is only allowed in "devDependencies". Please remove it from the "dependencies" section in your package.json.
    at Packager.checkMetadata (/Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/src/packager.ts:381:13)
    at /Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/src/packager.ts:169:10
    at Generator.next (<anonymous>)
    at runCallback (timers.js:800:20)
    at tryOnImmediate (timers.js:762:5)
    at processImmediate [as _immediateCallback] (timers.js:733:5)
From previous event:
    at Packager.build (/Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/out/packager.js:284:11)
    at /Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/src/builder.ts:198:33
    at Generator.next (<anonymous>)
From previous event:
    at build (/Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/out/builder.js:70:21)
    at Object.<anonymous> (/Users/DarklyBase/Projects/mybuild/lw.comm-server/node_modules/electron-builder/out/cli/build-cli.js:71:41)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Function.Module.runMain (module.js:605:10)
    at startup (bootstrap_node.js:158:16)
    at bootstrap_node.js:575:3
npm ERR! code ELIFECYCLE
npm ERR! errno 255
npm ERR! lw.comm-server@4.0.96 distmac: `npm run copy && build --mac `
npm ERR! Exit status 255
npm ERR! 
npm ERR! Failed at the lw.comm-server@4.0.96 distmac script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/DarklyBase/.npm/_logs/2017-06-18T22_24_47_098Z-debug.log
```

domenic-d, 2017-06-18:
Am I the only one having this problem?
@jorgerobles I know you compile on Mac, are you having the same issues?

sdesalas, 2017-06-19:
Hey guys, I'm getting an error with version `1.0.8` of `merge-jsons-webpack-plugin`, might be related, the breaking change went out 2 days ago (17 june).

```
ERROR in ./reducers/machine-profiles.js
Module not found: Error: Cannot resolve 'file' or 'directory' ../data/machine-profiles.json in E:\PROJECTS\LaserWeb4\src\reducers
 @ ./reducers/machine-profiles.js 20:76-116
```

sdesalas, 2017-06-19:
Added PR for this ☝️

sdesalas, 2017-06-19:
Also, there might be a bit of of a problem with [this commit](https://github.com/LaserWeb/lw.comm-server/commit/70ae8e1b8d7d942cc26766b98c366678aef89229#diff-b9cfc7f2cdf78a7f4b91a753d10865a2R40):

[image]

closed 2017-06-19 by jorgerobles

jorgerobles, 2017-06-19:
Sorry, I was away the weekend.
I've merged pull requests pending and fixed #35 with @sdsalas patch-
I was able to build on clean install.

cprezzi, 2017-06-19:
@sdesalas Thank you for the hint! I didn't realize that.
@jorgerobles Thank you for fixing!

sdesalas, 2017-06-19:
🕺 wohoo!!! 👍 👍

domenic-d, 2017-06-19:
No problems building OSX now.
Thanks everyone!

commit 047a706f (2017-06-19): FIx #35
commit f0028692 (2017-06-19): FIx #35