LaserWeb/lw.comm-server #38: Pi3 Build Problems

TechMasterJoe, 2017-06-17:
I did my best to get as much info as a can this error happens with node v6~8
at first i was getting stopped by serialport missing problem build rebuilding it before running install worked for that now just getting build run problems
laserweb is nice but the only platform that I want to run it on is a pi inside my K40 

[electron_bundle error.txt](https://github.com/LaserWeb/lw.comm-server/files/1082745/electron_bundle.error.txt)
[electron_bundle NPM log.txt](https://github.com/LaserWeb/lw.comm-server/files/1082746/electron_bundle.NPM.log.txt)

tbfleming, 2017-06-17:
Electron's only purpose is so end users can run a prebuilt installer instead of using git and npm commands. Do this instead:

```
git clone https://github.com/LaserWeb/LaserWeb4.git
cd LaserWeb4
npm run installdev
```

This starts the server:

```
cd LaserWeb4
npm start
```

Use Chrome 58 or later to connect.

cprezzi, 2017-06-18:
@tbfleming He doesn't need to clone LaserWeb4, as lw.comm-server serves the dist files from LW4 in the app folder!

@TechMasterJoe Just follow the description here: http://cncpro.co/index.php/23-documentation/installation/36-install-raspberry-pi

bullestock, 2017-06-30:
@cprezzi Those instructions do not work. `npm start`  simply outputs a message stating that you should run `npm run dist`. That does not work (after fixing a problem related to `process.arch`, it fails on `gm convert`). And `npm run installdev` fails with `missing script: installdev`.

cprezzi, 2017-07-02:
@bullestock Thanks for the hint. That's caused by a recent change of package.json for compatibillity with electron bundler. Just use `node server.js` to start the server.

closed 2017-07-02 by cprezzi