LaserWeb/lw.comm-server #39: feed overrides create OverLapped Results

openhardwarecoza, 2017-06-27:
1) Connect to Smoothie
2) Start a Job
3) Click the Feedrate Overides
3.1) Get pissed at 1% being too small an increment (make it +10% please - practically speaking i cant see why anyone would do 1% increments during a job. When shit starts chattering you want to adjust the feed FAST! you dont have time for hundreds of clicks. ) ) 
3.2) Loose connection to your machine because of the overlapped write, break endmill, ruin material. Realise its last material you had on hand.  fuck

cprezzi, 2017-06-27:
You can CTRL-Click to get 10% stepps (help showed on button mousover).
1% stepps are very usefull to find the sweet spot, especially with laser power.

cprezzi, 2017-06-27:
What do you mean with "create overlapped results"?

openhardwarecoza, 2017-06-27:
That works for me. (:

openhardwarecoza, 2017-06-27:
"What do you mean with "create overlapped results"?":  Happens when nodede-serialport is trying to write two streams to the port (queue and realtime) 

https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/src/serialport_win.cpp#L277

openhardwarecoza, 2017-06-27:
I think port.write has a callback when its done writing. perhaps push realtime in between that and next send1q?

openhardwarecoza, 2017-06-27:
PS: on my faster desktop it doesnt happen often. but on my cheap chinese windows 10 netbook, it happens almost every time i try using override, but always after a couple clicks, 10+

openhardwarecoza, 2017-06-27:
Or it could be that win7 works better, and win10 doesnt... dont have another win10 to test against

cprezzi, 2017-06-27:
I work with Win10 but never got this error related to overrides or other "real time" commands. 
It just happend sometimes mid job with smoothieware, but not anymore with grbl-LPC. Probably has something to do with different USB protocolls.

openhardwarecoza, 2017-07-04:
Okay, still happens on the new Asus laptop, so ruling out slower CPU being the main culprit. Did however notice it happening easier during short segment moves. Mentally i hold off feedrate adjustments till its about to do longer sections. 

Error code is 121

domenic-d, 2017-07-10:
We have just had a few reports from users that they are getting the overlapped results error as well.
[image]

Users are running 'Windows 10 home version 1607'
Is there any further development on this?
Is there any information that would be useful fur us to provide to help resolve this?

openhardwarecoza, 2017-07-11:
@domenic-d , also happened when using feed/power overrides?

domenic-d, 2017-07-11:
I checked with them after your feedback and no users reported making any changes to feed/power overrides.

openhardwarecoza, 2017-07-11:
Find out EXACTLY when / why they got the error.  It shouldnt happen during normal operations (as during normal operation we run just the single file queue (out, ok, out, ok).  Its during overrides that an extra line of commands are dumped onto the port...

openhardwarecoza, 2017-07-11:
(Maybe they dont know the up/down buttons on the DRO are called "feedrate override" and "S override" - so they report they didnt because they dont know what they clicked>?

cprezzi, 2017-07-11:
According to the screenshot, it's not a firmware-**cnc** version.
And what are those two Ellapsed lines? Is this a new special smoothie firmware?

@openhardwarecoza The overrides are not directly sent to the serial port, they just push the command in front of the queue instead of the end.

domenic-d, 2017-07-11:
@cprezzi This is the current version of Smoothiware. There is nothing special about it.
What classifies it as smoothiware-cnc on your system?

cprezzi, 2017-07-11:
Smoothieware should send a second line with CNC in it after the first line, but probably this is only visible at the server console. I will download and test the latest smoothieware later today.

cprezzi, 2017-07-11:
This is how it should look like when using the right firmware-cnc.bin:

[image]

cprezzi, 2017-07-11:
Please advise they should use this firmware: https://github.com/Smoothieware/Smoothieware/blob/edge/FirmwareBin/firmware-cnc.bin

cprezzi, 2017-07-11:
@domenic-d Can you provide a file and the settings which caused the OverlapedResult error?

cprezzi, 2017-07-11:
I was able to replicate the OverlappedResult error when using a cheap 1.8m USB cable, while it worked flawlessly with a good quality 1.8m cable or a standard blue 0.5m cable (without ferrites)!

All this was tested with the same front panel USB port on a desktop PC with Windows 10 and my C3D Remix.

Then I measured the USB voltage on the board side:
- Standard blue 0.5m cable: 4.3V
- Good 1.8m cable: 4.3V
- Cheap 1.8m cable: **4.0V** !

Changed to a rear USB port (onboard) I got 4.16V with the cheap 1.8m cable and the communication was stable.

It seems that not all USB ports provide the correct 5V and bad cables decrease that even more. If the voltage on the board dops below ~ 4.2V, the serial communication (and whatever else) seems to be unstable.

openhardwarecoza, 2017-07-11:
Thats why my boards, and Domenic's custom boards, neither rely on the USB
bus voltage. We have 5v regulators onboard (;

cprezzi, 2017-07-11:
Unfortunately most users with Smoothie or C3D boards do not have a 5V supply.

domenic-d, 2017-07-11:
@cprezzi @openhardwarecoza I am asking for more information. I'll pass it on as soon as I have it.

openhardwarecoza, 2017-07-11:
Indeed... But in the case of me and Domenic, the issue is not supply
voltage

openhardwarecoza, 2017-07-11:
C3D should also have proper supply. Its based off my Smoothiebrainz where i
already added the psu. Ray just swopped the LM2596 for a smaller XL1509. I
am using the XL1509 on my newer boards too.

cprezzi, 2017-07-11:
Ok. I was testing with the Cohesion Remix but without external power connected. If I connect the external 12V supply, the board voltage increases to 4.8V and it works also with the cheap 1.8m cable.

domenic-d, 2017-07-12:
We have been able to replicate 'our' specific problem on our machines in the lab.
It is caused by plugging the smoothieboard into a USB3 or SSUSB port.
After a short period of time, the firmware crashes and these errors are reported. 
It is not related to Laserweb.

We will take this up with the Smoothie team.

openhardwarecoza, 2017-07-12:
Hmmm, interesting observation.  I'll check if the old Asus i stole from my
wife (which she got new like 5 years ago) even has USB3 (; - i think its
too old (;

cprezzi, 2017-07-12:
My PC also hasn't USB3, but one of the ports is combined USB/eSATA, and this port doesn't work with the Remix board.

domenic-d, 2017-07-12:
It is an unusual problem which we have been testing all day. We have tested the latest smoothiware firmware (both cnc and reprap). Both behave the same, with the firmware crash.
This all became apparent with the new laptop we purchased for our QA workstation. None of our other older computers can produce this problem.

openhardwarecoza, 2017-07-12:
I am outside in the workshop now. Checked and low and behold this laptop
does have usb3. I am not sure if thats where i was plugged into... Let me
try deliberately using a usb2 port this afternoon (got some lasering todo
for the morning and thats on a Grbl based board. Will hook up the mill
(which runs smoothie) this afternoon and compare. Looks like you are maybe
onto something Dom!

openhardwarecoza, 2017-07-18:
Okay. So running on the usb2 port for the last couple days have indeed proven to be more stable. But not eliminated the issue. Now, i still get getOverlapped-121 errors, but, they happen when i have feedrate override over 180% and, then it happens randomly mid job (not only on fro clicks). Starting to think grbl-lpc, suspect Smoothieware or their usb driver, again, at blame