LaserWeb/lw.comm-server #41: Home Z crash server

jorgerobles, 2017-07-04:
Just did Home Z :cry:

[image]

cprezzi, 2017-07-04:
Did you see the `Smoothieware detected (...)` string in the console? 
Could you please test if `G28.2 Z` is working correctly of delivering the same error?

jorgerobles, 2017-07-04:
[image]

G28.2 Z Works fine

tbfleming, 2017-07-04:
My heart skipped a beat when I read the first 3 words of the title: "Home Z crash"

cprezzi, 2017-07-04:
Ok, found the error. Typo in the firmware name (smothie insted of smoothie). Fixed with commit https://github.com/LaserWeb/lw.comm-server/commit/cc3741595ecd5e4f49fec4e25b5f1303e71859e8

jorgerobles, 2017-07-04:
> smoothie). Fixed with commit cc37415

Great!!

closed 2017-07-06 by jorgerobles