LaserWeb/lw.comm-server #42: Fix logfile path

jorgerobles, 2017-07-06:
As https://github.com/LaserWeb/LaserWeb4/issues/369 @gerwinbrunner

@cprezzi I've tested patch on `electron_bundler` 
At https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/server.js#L2843

`logFile = fs.createWriteStream(path.join(electronApp.getPath('userData'),"logfile.log"));`

I've no commited this, as don't know if thas to be other considerations apart electron

cprezzi, 2017-07-06:
Yes, that's the problem! If we fix that only in the electron_bundler branch, it will probably be overwritten by a merge with master, but I can't do electron specific stuff in master.

I think this is a bug in electron for iOS, as a filename without path should use the default path which is the installation path.

cprezzi, 2017-07-06:
Is there something like electron compile switches we could use?

jorgerobles, 2017-07-06:
I've not found anything on `electron-builder`

What I would do:

1. Move all the electron related code to other file (app.js ?)
2. Encapsulate server code on an class (object/prototype/singleton, or even nested function) and inject dependencies on construct (like a `setLogger` function that swallows a modified `logFile` function)
3. Create a new bootstrap.js file, make it main at package.json.
4. Bootstrap.js reads an `.env` file git-ignored in the repo, if not exists instances the `server.js` class and runs. If exists, load the alternative file on env (like `app.js`), that runs the server.js instance and the electron part.

tbfleming, 2017-07-06:
Maybe ```if(electronApp)```?

cprezzi, 2017-07-06:
@jorgerobles Isn't that a bit overcomplicated? There must be a simpler solution.

cprezzi, 2017-07-06:
Ok, found something to get if app is running inside electron: 
`if (process.versions.electron) {...`

jorgerobles, 2017-07-06:
Well is complicated but abstracts enough to update server freely.  Proces.. is of course valid

cheton, 2017-07-06:
You can also use [is-electron](https://github.com/cheton/is-electron) to detect if current process is running in Electron.

cprezzi, 2017-07-06:
Thank's @cheton! That's even better.

cprezzi, 2017-07-06:
@jorgerobles `logFile = fs.createWriteStream(path.join(electronApp.getPath('userData'),"logfile.log"));` doesn't seem to work on Win10 electron app. I get now error message, but also no logfile.

jorgerobles, 2017-07-06:
Probably is working but saving somwhere else :)

cprezzi, 2017-07-06:
Nope, doesn't seem to. I searched the whole drive. 
Looks like we need a way to find out if running on Mac 👎 (hate that stuff!)

tbfleming, 2017-07-06:
If(Mac) warn("update to an OS with sane APIs")

cprezzi, 2017-07-06:
Ok, found that getPath('userData') is delivering the correct path with \ as directory separator on Windows. Still searching why fs.createWriteStream is not working...

cprezzi, 2017-07-07:
I had to make if OS dependant with `os.platform`.

Who could compile the Mac Version and test if the logfile is written now? (needs .env file with LOG_LEVEL=3 in installation folder).

jorgerobles, 2017-07-07:
I'll do

jorgerobles, 2017-07-07:
[image]

But does not write log :|

jorgerobles, 2017-07-07:
@cprezzi  cannot get it work on Sierra. Does not find .env

cprezzi, 2017-07-07:
I don't know anything about Mac. On Windows, the .env file just has to sit in the same directory as the exe.

jorgerobles, 2017-07-07:
Yes. I've tried several ways. Seems version bound. On Mavericks ran ok yesterday.

cprezzi, 2017-07-07:
Eventually this could help: https://apple.stackexchange.com/questions/106778/how-do-i-set-environment-variables-on-os-x

jorgerobles, 2017-07-07:
Yes, I did the `export LOG_LEVEL=3` too 😢

cprezzi, 2017-07-07:
Did I say - I hate Apple?!

jorgerobles, 2017-07-07:
Hmm. Osx could contribute, but I see more culprit on electron or dotenv. I
don't miss win viruses.

cojarbi, 2017-07-07:
Osx is a God gift. :)

tbfleming, 2017-07-07:
I once gave a MBP 3 months of my development life. I liked the hardware so was willing to give the OS a fair shot. Things kept getting worse over time as I learned more and more of it. It's both a bad UI (very horrid keyboard support, way too much clicky needed) and a bad POSIX system (ancient BSD version missing critical syscalls needed to avoid deadlock in fork/exec).

DouglasPearless, 2017-08-22:
This is how I got the Logfile.txt working on OSX:

Open a Terminal session (from the Utilities folder under Applications)
`cd /Applications/Laserweb.app/Contents`
`export LOG_LEVEL=3`
`./Laserweb`

And as it runs it creates Logfile.txt in that same directory.

Cheers
Douglas.

cprezzi, 2017-08-22:
@DouglasPearless Thank you for the information. That makes sense. Setting the environment variable with export only survives in the same terminal session.