LaserWeb/lw.comm-server #43: Z work offsets incrementing with homing

domenic-d, 2017-07-18:
Z work offset keep incrementing by 10mm every time a homing is performed.

[image]

cprezzi, 2017-07-19:
It seems you are not using the firmware-cnc right? Otherwise there should be a second line after the Firmware detection saying `CNC Build 3 axis`.

cprezzi, 2017-07-19:
[image]

cprezzi, 2017-07-19:
Unfortunately I do not have a machine with Z-Axis and smoothie at the moment, so I can't test myself.

cprezzi, 2017-07-19:
The firmware in your screenshot is firmware.bin from 27 Nov 2016! 
Please try [firmware-cnc.bin](https://github.com/Smoothieware/Smoothieware/blob/edge/FirmwareBin/firmware-cnc.bin) from 24 Apr 2017 as recommended.

domenic-d, 2017-07-19:
We are using this version because it has been working correctly for our machine.  In case you were not aware, updating the firmware for a certified product, such as ours, is not a straight forward process. There are many organisations to be notified and paperwork that needs to be officially generated.

The cnc version of the firmware also makes a significant change to the way G28 is handled and this would cause all our existing customer's machines to stop functioning correctly unless they all performed the updates. 

Having said all this, we will have to work through this and resolve it at our end. It is unfortunate that consideration for legacy support was not made when this change was made.

tbfleming, 2017-07-19:
LW4 was a fresh start. If I remember right, we've always said LW4 needed the -cnc version.

domenic-d, 2017-07-20:
Installed the latest smoothie 
firmware CNC version.
Problem still exists.

[image]

cprezzi, 2017-07-20:
@domenic-d There was issues in smoothieware that have been solved since last november and some new features was added, that we use too. Did you realy plan to stay with this version for Years?

The LaserWeb Porject needs to stay up-to-date with the latest firmwares and often we depend on new firmware features.

cprezzi, 2017-07-20:
@domenic-d Now to your problem: 
- Could you please provide your smoothie config file?
- Please provide your LW settings and machine profiles (export from settings/tools)?
- How is your mechanical endstop configuration (microswitch/optical, NO/NC, to GND/5V)? 

@openhardwarecoza @cojarbi Could you please try to replicate this behaviour?

openhardwarecoza, 2017-07-20:
No homing switches installed. Sorry (:

cprezzi, 2017-07-20:
@domenic-d 
- Which OS and LW version did you use? 
- Did you test this with a USB cable or/and your bluethooth bridge? (any difference?)

cojarbi, 2017-07-20:
Me neither. I will prep a machine soon

domenic-d, 2017-07-20:
We have tested this further and found that the homing commands we were using was causing the problem. With the CNC version of the firmware, the G28.X commands must be used to ensure this issue doesn't arise.
For reference in case others have this problem:

**Original Homing Sequence**
G0Z60
G92Z50
G28X0Y0

**Revised homing sequence**
G0Z60
G28.3Z50
G28.2XY

Apologies for this false alarm and thank you for the help.

closed 2017-07-20 by domenic-d

openhardwarecoza, 2017-07-21:
Remember to update your lw. Machines profiles so your profiles get included