LaserWeb/lw.comm-server #44: More elegantly handle smoothieware "reset"

openhardwarecoza, 2017-07-31:
[image]

Upon issueing the "reset" command, it will always happen that Smoothieware closes the port (since its USB stack is in software, not like with Grbl where mostly its on a seperate ftdi/ch340/silabs).  However, now a) LW keeps trying to talk to the port, and b) sometimes keeps the port open in windows so the subsequent re-enumeration fails. 

In this case, we now need to switch to comms tab, disconnect, then manually replug the USB to get enumeration after reset to work.  

Resetting from software is convenient during that initial machine setup / after certain console commands

We can extend this functionality to a proper "reset" ws-event if you like?  Then we can have:

-  A 'reset controller' button on comms tab, just like pronterface has always had
- switch case to handle firmware differences ('reset' for smoothie + disconnect port,   for grbl its ctrl+x (ascii \030 aka port.write("\030"), etc)

cprezzi, 2017-08-02:
I do already watch for the close event of a com port, which does a cleanup, but I don't know why this event doesn't fire when smoothie closes the port.
And there is already a *resetMachine* command in lw.comm-server which does the firmware specific stuff, but at the moment it sends ctrl-x (0x18) to grbl, smoothie and tinyg (or M112 to marlinkimbra). I thought smoothie supports ctrl-x?

tbfleming, 2017-08-02:
The USB CDC protocol doesn't define open and close; the OS provides those abstractions. Under Windows, only the application can close a port. If a device is unplugged (smoothie reset kind-of looks like this to the host), then the file handle enters an error state.

cprezzi, 2017-08-02:
I do also catch error events, but not getting it.

cprezzi, 2017-08-02:
@tbfleming But why does a software reset behave different than a hardware reset (button)? Smoothie should be able to handle that the same way, especially because the USB driver is part of the code.

tbfleming, 2017-08-02:
I don't know. ctrl-x in grbl just resets some, not all, state, which allowed me to make it so the PC sees no funny business on the USB bus.

openhardwarecoza, 2017-08-03:
I did some more testing... After sending reset, Smoothie does the 5 second countdown before resetting.  If i do Disconnect from comms tab before it resets, i have not had a problem re-enumerating yet, did about 20-25 runs of this.  

@cprezzi ""I do already watch for the close event of a com port" - is this event on WS or SerialPort? If its on SP, perhaps still react to the WS event too?  Ie close the port from node (ie serialport port close) as soon as that "reset" (or Ctrl+X if that resets it too) has been sent if firmware==smoothie?

openhardwarecoza, 2017-08-03:
resetMachine command in lw.comm-server - can we get a Reset button on the comms tab (;

cprezzi, 2017-08-04:
I think the jog tab will get some rework anyways. Todd is already working on configurable forms. :)