LaserWeb/lw.comm-server #45: Unespecified error relating consumed time

jorgerobles, 2017-08-02:
@cprezzi Sorry for being so vague. Was doing a cut with latest github source, and suddenly lw.comm-server stopped working, throwing an error related gcode length (I closed accidentally the log)
The problem is not arised with the latest binary release. :/

cprezzi, 2017-08-02:
That sounds more like a error message that came from the firmware after trying to send a too long gcode line (probably caused by unstable connection). The job time calculation is only listening to status messages and not sending any data to the machine.

What firmware and controller do you use?

jorgerobles, 2017-08-02:
Grbl lpc, the one you help me to install a couple of months ago

cprezzi, 2017-08-02:
Would be helpfull to reproduce the error or at least have some more infos.

jorgerobles, 2017-08-02:
I know ;) will try to reproduce

closed 2019-02-14 by cprezzi