LaserWeb/lw.comm-server #48: Suggestion: Laser Fill Path in one pass

jameswood, 2017-08-22:
On plastic materials, I have better results using Laser Raster mode than I do with Laser Fill Path. This is because Laser Raster mode starts at the bottom and scans up the image line-by-line in one pass. Laser Fill Path does not behave this way; instead, it fills areas piece by piece. This causes lines to appear at the junctions of the various "start" and "stop" points, because the material has cooled while other areas are engraved.

I'd love an option within Laser Fill Path that makes it behave more like Raster mode - start at the bottom and scan to the top in one pass.

tbfleming, 2017-08-22:
Use Laser Raster Merge

jameswood, 2017-08-23:
Hi Todd, thanks for your reply.

Two points:
1. I didn't realise how old my frontend version was... or how quickly you release versions...
2. you're awesome for already having built the thing I thought I was suggesting.

Sorry for wasting your time and thanks for your help!

closed 2017-08-23 by jameswood