LaserWeb/lw.comm-server #49: Installing lw.comm-server on OrangePi Zero Armbian xeniel_default 3.4.113

techpad-marcb, 2017-09-13:
Just wanted to log a place for install steps for Armbian 3.4.113 for lw.comm-server

**Clone Git**
`sudo git clone https://github.com/LaserWeb/lw.comm-server.git`
`cd lw.comm.server`

Copy over missing files after clone to lw.comm-server folder
config.js, firmwareFeatures.js, grblStrings.js, lw.comm-server.service, server.js, package.json

**Install dependencies first**
`sudo npm install serialport --unsafe-perm --build-from-source`
`sudo npm install dotenv --unsafe-perm --build-from-source`
`sudo npm install debug --unsafe-perm --build-from-source`
`sudo npm install ajv-keywords --unsafe-perm --build-from-source`
`sudo npm install 7zip-bin --unsafe-perm --build-from-source`

`npm install`

Got this far and server loads but unable to connect to it from web interface.

***************************************************************
        ---- LaserWeb Comm Server 4.0.115 ----
***************************************************************
  Use  http://127.0.0.1:8000 to connect this server.

* Updates:
  Remember to check the commit log on
  https://github.com/LaserWeb/lw.comm-server/commits/master
  regularly, to know about updates and fixes, and then when ready
  update accordingly by running git pull

* Support:
  If you need help / support, come over to
  https://plus.google.com/communities/115879488566665599508
***************************************************************

I feel like it has something to do with the IP address being 127.0.0.1 but i don't know more tinkering to come i guess.

cprezzi, 2017-09-13:
You need to use the real IP address of your Raspberry Pi to connect it from the network. 127.0.0.1 is only available locally from the Raspi itself. Check `/sbin/ifconfig -A` for your IP.

techpad-marcb, 2017-09-13:
Now when I connect to the server I get this error:
/home/*******/lw.comm-server/node_modules/node-static/lib/node-static.js:348
                      mime.lookup(files[0]) ||
                           ^

TypeError: mime.lookup is not a function
    at Server.respond (/home/*******/lw.comm-server/node_modules/node-static/lib/node-static.js:348:28)
    at /home/********/lw.comm-server/node_modules/node-static/lib/node-static.js:64:22
    at FSReqWrap.oncomplete (fs.js:114:15)

cprezzi, 2017-09-13:
Seems like some version conflicts. Did you follow the instructions from the Wiki exactly? 
We need NodeJS LTS Version 6.11, not the Current Version 8.5!

techpad-marcb, 2017-09-13:
I have version 7.10.1 would that cause an issue?

tbfleming, 2017-09-13:
@techpad-marcb We have to coordinate version bumps between Node and Electron. We don't do it often. As far as I know no one's tested 7, so consider it bad.

tbfleming, 2017-09-13:
Side note: We don't have a strong incentive to update Electron for client-side development until Electron reaches Chromium 60.

techpad-marcb, 2017-09-13:
Ok rolling back to 6.1.1 and trying again. Sorry new to this stuff just installed nodejs and didn't think anything of it.

techpad-marcb, 2017-09-13:
Nope still the same error with mime.lookup so i'm at a loss here no idea what to do next. Would it matter that it is 6.11.3? Followed the install instructions and thats the version i got.

tbfleming, 2017-09-13:
The .3 indicates just a bug-fix release, so that's probably not the problem. Did you wipe node_modules? That should be wiped whenever you switch node versions.

techpad-marcb, 2017-09-13:
I think so deleted the entire lw.comm-server folder and started over after removing nodejs

Getting this now after a reboot 

***************************************************************
        ---- LaserWeb Comm Server 4.0.115 ----
***************************************************************
  Use  http://127.0.0.1:8000 to connect this server.

* Updates:
  Remember to check the commit log on
  https://github.com/LaserWeb/lw.comm-server/commits/master
  regularly, to know about updates and fixes, and then when ready
  update accordingly by running git pull

* Support:
  If you need help / support, come over to
  https://plus.google.com/communities/115879488566665599508
***************************************************************

ERROR:  webServer error:/ :  Not Found

Connecting from my desktop to 192.168.0.103:8000

techpad-marcb, 2017-09-13:
At this point this is where i installed electron thinking that for some reason the server was not installed to serve web pages.

cprezzi, 2017-09-13:
The response looks like it should. Could you ping the OrangePi from another PC and connect to SSH (if activated)?

techpad-marcb, 2017-09-13:
Yep I can ssh no problem that is how I have been configuring it. Octoprint works fine as well too on port 5000

techpad-marcb, 2017-09-15:
Any other suggestions?

techpad-marcb, 2017-09-15:
Ok i'm a big dummy. Installed laserweb app on my desktop and was able to connect and control. 

I totally thought that the server was a standalone application that would display a web interface like the thick client. All good thanks.

So the default instructions to install worked perfect.

closed 2017-09-15 by cprezzi