LaserWeb/lw.comm-server #53: 250k Baud

reconbot, 2018-02-06:
As of `serialport@6.1.0` we now have custom buadrate support for linux. This will allow buadrates on linux of 250k and maybe even ones that are faster. I'd love if people here could give it a try and let me know if it work for them. I'm most curious about all the flavors of linux that run on embedded devices.

This would also solve some of the windows issues your windows users are undoubtedly having and some memory leaks on linux and OS X.

tbfleming, 2018-02-06:
Most of our users are on smoothieboards, which ignore the baud rate in their CDC implementation. I'm curious if it will help stability.

@cprezzi

cprezzi, 2018-02-06:
@reconbot Thanks for the hint! Do we have to expect ane incompatibilities?

@tbfleming We should test the new serialport version before pushing it to the release.

reconbot, 2018-02-06:
Upgrading from 4 there are a few changes, disconnects are now part of the close event. We're now a stream. Some options are a bit different. You can see our upgrade guide in the repo.

reconbot, 2018-02-06:
I'm happy to review any prs, doesn't look it it would be any Major changes

cprezzi, 2019-02-14:
Closed as we moved to Serialport 6.2.2

closed 2019-02-14 by cprezzi