LaserWeb/lw.comm-server #57: Updates to improve docker compatibility

gdunstone, 2018-05-17:
firmwareFeatures.js is not added in the dockerfile, this causes the app not to start.

change to *Dockerfile:3*

`ADD config.js firmwareFeatures.js grblStrings.js LICENSE lw.comm-server.service package.json README.md server.js version.txt /laserweb/`

Also npm start doesnt seem to work and is actually not even used in the systemd unit.

change to *docker_entrypoint.sh:5*

`nice -n -20 node server.js`

Also it would be good to have some info on how to build and run the docker image in the wiki. Just some stuff like using `--cap-add=SYS_NICE` and `--device=/dev/ttyUSB0`

cprezzi, 2018-05-17:
You are welcome to send a pull request.
I'm a docker noob. Why should someone want to run LW as docker on a raspi, if the app is running natively?

jorgerobles, 2018-05-17:
Docker creates a virtualenv encapsulating the requirements (like having all
packaged on a blackbox) It's very useful, and should not eat much resources
doing so in the same machine is emulating :)

gdunstone, 2018-05-17:
I'll make a pull request later today then.