LaserWeb/lw.comm-server #60: APP folder (UI) is too old. Server-UI decoupling

jorgerobles, 2018-05-22:
Hi,

I've run on a somewhat serious issue due the folder App being far away current version (8 months!)
I've been refactoring the binaries build chain, decoupling lw.comm-server from electron (at least for WIN and OSX, cannot get LNX working on multiple travis file) 
Then the version bundled was way back because nstatic was serving the inner lw.comm-server app instead the new UI (I've patched the APP using travis and overriding the folder with the newer)

I would suggest to not to include APP on the comms, but make a build script to download and compile first time if folder does not exist (for users not using the bundled version)  
For example using travis to automatically build the app folder and being able to download from Github Laserweb4 Releases (not binaries, just LW4 containing the app.zip)

Also, I would like to have the option to not to run `nstatic` or be able to change the path where is served, the builder can choose another app folder. That could lead to having an UI updater without changing nor the builder nor the server.

@cprezzi, @tbfleming

cprezzi, 2018-05-23:
My idea was to pack a responsive mini frontend into the static app folder. This could be usefull to controll the machine with a tablett or mobile and should not interfer with the installed LW Application.

jorgerobles, 2018-05-23:
I've managed to upload ui dist automatically to lw4 releases as ui.asar and
to gh-pages.

Server part just need to download the ui.asar and unpack to /app upon
installation, or run as packaged with asar lib This way is always fresh and
updated with the latest version. Could even add an --update option to
redownload the ui

The other thing is to have a way to not to serve with nstatic, letting the
electron app deal with the ui.