LaserWeb/lw.comm-server #62: FEATURE: API for saving/loading settings

quintesse, 2018-06-23:
This would be the server part of https://github.com/LaserWeb/LaserWeb4/issues/49

A way for the UI to store and retrieve settings from the server. This way the user could easily set up different clients because the settings would already be available on the server that it connects to.

quintesse, 2018-06-27:
@cprezzi understood, I had just wanted to keep it simpler, but for people with multiple machines connected to a single server a single profile would not work indeed. Thanks for the feedback!

Gleapsite, 2019-02-15:
>  there are users with multiple different machines that would need different profiles.

Meaning, one instance of lw.comm-server that can communicate with multiple machines.  

Wouldn't this feature be similar to how the material database is handled?

cprezzi, 2019-02-18:
@Gleapsite If you connect multiple machines to the PC running lw.comm-server, then you can select in the frontend, which machine (port) to connect. 
For example: I run lw.comm-server on a Raspberry Pi 3B+ and have connected two machines to it via USB. I can then decide which machine I want to work with by loading the matching machine porfile and connect the right USB port.

By the way, it's also possible to connect to lw.comm-server with more than one client at the same time. For example the PC to create the gcode and run the job and a tablet to watch and pause/resume the job.

quintesse, 2019-02-18:
> By the way, it's also possible to connect to lw.comm-server with more than one client at the same time. For example the PC to create the gcode and run the job and a tablet to watch and pause/resume the job.

@cprezzi it is? Is that new-ish functionality perhaps? Because I remember trying this many months ago and was surprised that I basically had to do the entire setup for each client and that each of them would see a different "state", so to speak. (Except for the actual physical state of the machine like laser position IIRC)

DataWorm, 2019-02-18:
Yeah same here! Would love such a feature, the lack of globally stored settings is the main show-stopper for me. Octoprint fits more to what I expect except that it is more focussed on 3d printing. The lasercutter features of LaserWeb integrated in Octoprint as plugins would be awesome but LaserWeb by itself would be good enough with central storage! =)

quintesse, 2019-02-18:
@DataWorm indeed, I use OctoPrint for my 3d printer and I love that I can access it in the exact same way from different clients. Something like that for a laser cutter would be excellent.

cprezzi, 2019-02-18:
The machine profiles are still stored on client side and need to be copied manually to additional clients. What I said is that the server can handle multiple socket connections at the same time.

cprezzi, 2019-02-20:
@quintesse You can also install lw-comm-server on a Raspberry Pi and access it from different clients (Win, Mac or Linux) either from a LaserWeb executable or just by a web browser. Only the profile that is used for the integrated gcode generation is stored on the client (only needs to be set up once).

What we don't have is the job storage on server side, so you could not upload gcode-files and run them later from a different client. But you upload gcode generated by LW4 and upload it OctoPrint, so you could run it later. It should not matter for OctoPrint, if the gcode file is for a 3d printer or a laser.

quintesse, 2019-02-20:
@cprezzi yes, but that setup is pretty complex. Each time I use a new client (or a client that has its cache cleared) I need to put it side-by-side with a working client and carefully copy all the settings. Even exporting/importing is not trivial when your new client is a tablet or something (export, mail, save, import).

And my LW is already running on a RPi, so it would be a bit weird to use that to generate gcode, download it and then upload it to OctoPrint to start cutting.

I know, it's just the way things are, but personally I just wish LW worked more like OctoPrint :-)

cprezzi, 2019-02-27:
@quintesse Yes, I understand that for your scenario it would be much easyer if the server had the profiles.
This is also the way we work with the lasersaur in our FabLab. People just open the server webpage with their devices, upload the files and then go to the PC aside the laser to run the job.

Maybe I will implement that server profile storage in the future, when I get some more time ;)

Gleapsite, 2019-02-28:
Yes. In my ideal use case,  I could connect to any given machine from an arbitrary client, and have that given machine's settings, job history, and materials database.   

To me, thats machine-specific data, as opposed to client specific.  

@cprezzi thank you for all the time you've put into this project already.  I wish I could help, but I'm not very fluent in nodejs