LaserWeb/lw.comm-server #64: NPM version issue and Serial port 4.0.7 missing?

175345, 2018-09-02:
Hello! 

I´m trying to install the lw.comm-server on my RPi 3B but are having some issues.

When following the guide on the Git page for manual install ( 

https://github.com/LaserWeb/lw.comm-server/wiki/Manual-Installation-(RasPi)  

I get the following issues:

**Issue #1 **
when installing Node 6.x I do not get the most up to date npm version.
The version I get is v1.4.21, and that is far to old as stated in the guide.
I found the solution for this I think on the following page:

https://stackoverflow.com/questions/44095985/why-can-i-only-install-an-outdated-version-of-npm-on-debian-ubuntu

So I started with a fresh copy of the Rasbian 4.14.50-v7+ again and this time issued the command:

```
curl -sL https://deb.nodesource.com/setup_8.x | sudo -E bash -
sudo apt-get install -y nodejs
```

This worked and the current version installed is:
pi@LaserWeb:~ $ npm -v
5.6.0
pi@LaserWeb:~ $ node -v
v8.11.4

All good for now right?

**Well, lets see Issue #2:**

Issue #2. 

Git cloning the lw.comm-server.git was no problem, but what comes after is a problem i think:

`cd /home/pi`
`git clone https://github.com/LaserWeb/lw.comm-server.git`
`cd lw.comm-server`
`npm install`

```
npm WARN deprecated hoek@2.16.3: The major version is no longer supported. Please update to 4.x or newer
npm WARN notice [SECURITY] debug has the following vulnerability: 1 low. Go here for more details: https://nodesecurity.io/advisories?
search=debug&version=2.3.3 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.
npm WARN notice [SECURITY] debug has the following vulnerability: 1 low. Go here for more details: https://nodesecurity.io/advisories?search=debug&version=2.2.0 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.
npm WARN notice [SECURITY] debug has the following vulnerability: 1 low. Go here for more details: https://nodesecurity.io/advisories?search=debug&version=2.3.3 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.
npm WARN notice [SECURITY] debug has the following vulnerability: 1 low. Go here for more details: https://nodesecurity.io/advisories?search=debug&version=2.3.3 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.
npm WARN notice [SECURITY] parsejson has the following vulnerability: 1 high. Go here for more details: https://nodesecurity.io/advisories?search=parsejson&version=0.0.3 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.
npm WARN notice [SECURITY] tunnel-agent has the following vulnerability: 1 moderate. Go here for more details: https://nodesecurity.io/advisories?search=tunnel-agent&version=0.4.3 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.
npm WARN notice [SECURITY] debug has the following vulnerability: 1 low. Go here for more details: https://nodesecurity.io/advisories?search=debug&version=2.6.1 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.
npm WARN notice [SECURITY] debug has the following vulnerability: 1 low. Go here for more details: https://nodesecurity.io/advisories?search=debug&version=2.6.1 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.

> serialport@4.0.7 install /home/pi/lw.comm-server/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/node',
node-pre-gyp verb cli   '/home/pi/lw.comm-server/node_modules/serialport/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@8.11.4 | linux | arm
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/home/pi/lw.comm-

server/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@8.11.4 (node-v57 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading 
tarballhttps://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz
 
node-pre-gyp verb command build [ 'rebuild' ]
make: Entering directory '/home/pi/lw.comm-server/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function 'int getIntFromObject(v8::Local<v8::Object>, std::__cxx11::string':
../src/serialport.cpp:90:52: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated: Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)>ToInt32()>Int32Value();
```

So, it cannot install the serial port since the git does not exist.

Any ideas?
I´ve tried to install mulitiple times with a fresh rasbian image, but with no luck.
Starting to lose my mind/patience.

Thanks in advance.

cprezzi, 2018-09-04:
We disussed this alredy on G+!
I guess it has something to do with Raspbian Stretch and sugest to use Raspbian Jessie.

cprezzi, 2018-09-17:
The serialport 6.2.2 version has been moved to a new branch "Serialport_6.2.2", because it breaks the compilation of the electron app (still works for Raspi users).

closed 2019-02-14 by cprezzi