LaserWeb/lw.comm-server #71: FEATURE: homing per axis on TinyG (g2core)

turboproc, 2019-01-19:
Would it be possible to add the homing per axis for TinyG (g2core) firmware. Just check it and homing per axis on g2core works the same way as it does in Marlin and Repetier.

cprezzi, 2019-01-19:
Homing for TinyG is not yet implemented, but sending homing commands like `G28.2 X0 Y0` with the console should work.

According to https://github.com/synthetos/g2/wiki/Homing-g2core homing does NOT work exactly the same as on Marlin or Repetier. The TinyG homing command is `G28.2 X0 Y0 Z0` where Marlin uses `G28 X Y Z`!

I will add support for TinyG homing to the next release.

turboproc, 2019-02-12:
Great, thnx. Just noticed. Building now myself, which brings some challanges, but should be done now. Testing soon.