LaserWeb/lw.comm-server #72: Building LW package

turboproc, 2019-02-14:
I'm trying to build a LW package for a WIN x64 environment on a Linux Ubuntu  environment, so cross environment. From a process point of view this work, *.exe file is created but running this in the WIN environment structurally results in an error message like "not a valid win32 app".

Could someone provide accurate building instructions to make this work?

cprezzi, 2019-02-14:
We always built the Windows versions on Windows. You sould try that first.

turboproc, 2019-02-17:
Got some interesting insights:
- building on linux targetted for Win32 requires the serialport@6.2.2 binaries to be available. Downgrading to 6.2.1 solves at least part of the problems.
- the resulting built seems to start the backend only. The frontend, webkit part, is not started so LW needs to be accessed via Chrome or alternarive.
- connecting to tinyg fails (firmware not recognized). When using one of the prefab builts this work flawles, so this issue must be introduced somewhere in the built process (or the Master branch is not accurate).

To find a way out, I created a Win10 environment. Need to install Visual Studio as well otherwise the native modules cannot be compiled. Results:
- the missing renderer (webkit) remains missing
- firmware recognition remains an issue

Would like to have some more details on the environment used to built releases and would like to understand whether the Master branch should be able to run.

Thnx.

turboproc, 2019-02-18:
Further update:
- firmware recognition has been solved. Had to include in server.js a port.set({cts:true}) line. Without this line the tinyg would not respond.
So now the webkit part remains but LW can still be used through a separate browser hence I can start to work on some changes and build the result myself for verification and before contributing.

cprezzi, 2019-07-29:
@turboproc It is correct, that lw.comm-server master branch doesn't start the frontent, because it's only ment to be used on a Raspberry Pi without frontend (with frontend on a network PC). 

If you want to compile the complete installable application, you need to use the electron_bundler branch. Look at the scripts in package.json.

closed 2019-10-08 by cprezzi