LaserWeb/lw.comm-server #73: implementing jogCancel command for live jogging

ibrahimuslu, 2019-03-01:
Hello, 

for the live jogging grblv1.1  implemented a jogcancel command with extended ascii commands
I have implemented it for arduino grblv1.1 but i dont know which commands for the other firmwares. 

```
      appSocket.on('jogCancel', function () {
        writeLog(chalk.red('Jog Cancel'), 1);
        if (isConnected) {
            writeLog('Jog canceled and queue cleared', 1);
            switch (firmware) {
            case 'grbl':
                addQ('85'.match(/.{1,2}/g).map(function(v){ // 0x85 hex jog_cancel command of grbl
                    return String.fromCharCode(parseInt(v, 16)); // converting ascii equivalent value 
                    }).join(''));  
                send1Q();
                break;
            default:
                writeLog(chalk.red('ERROR: ') + chalk.blue('Unknown firmware!'), 1);
                break;
            }
        } else {
            io.sockets.emit("connectStatus", 'closed');
            io.sockets.emit('connectStatus', 'Connect');
            writeLog(chalk.red('ERROR: ') + chalk.blue('Machine connection not open!'), 1);
        }
    });
```

cprezzi, 2019-03-01:
I don't see the need for "jogCancel". Why not just use "resetMachine"? This can be quickly issued by pressing ctrl-x from the frontend.

ibrahimuslu, 2019-03-01:
Let me explain. 
We are trying to use in lw.com.server and developing a new ui. jogcancel is stopping immediately the jog operation. there is no ctrl+x on touch screen. 
on the live jogging we are trying to implement jogging while pressing the button and stop jogging when you stop pressing. There is no other option that i could find to stop jogging immedieately and enter to idle mod.