LaserWeb/lw.comm-server #74: raspi install correction

pen25, 2019-03-22:
if running a pi3 and armv7 i had to use 

curl -sL http://deb.nodesource.com/setup_10.x |sudo bash -

sudo apt-get install nodejs

sudo chown -R $(whoami) ~/.npm
sudo chown -R $(whoami) /usr/local/lib/node_modules

cd lw.comm-server

npm install

lackybee, 2019-04-11:
I'm also having issues with raspberry pi install but lack the knowledge to work around them. The correct version of npm doesn't install when following the wiki instructions.

Following pen25's instructions I have access denied errors even when using sudo npm install

pen25, 2019-04-12:
you did these correct?

sudo chown -R $(whoami) ~/.npm
sudo chown -R $(whoami) /usr/local/lib/node_modules

lackybee, 2019-04-12:
I did finally get it working actually. I started again and used your curl command from above to get the newer nodejs and then followed the instructions on the wiki. I didn't need the change ownership commands above. My first attempt was on a raspbian lite install, second time around i started with the full desktop version.
Not sure what made the difference but I can say with some certainty that using node 6 doesn't work, it seems to have a lower version of npm bundled with it.
Now I have server connected but not the machine, thats a different problem though.

pen25, 2019-04-12:
Cpu dependant possibly? Either way you are up and running.

lackybee, 2019-04-12:
Yep, now that I have the steps nailed down I might try again on the lite OS

pen25, 2019-04-12:
Let us know.

cprezzi, 2019-04-12:
@lackybee So does this mean it did work when you just install node v10 (instead of v6) and then follow the rest of the wiki instructions?

pen25, 2019-04-13:
I can say v6 would not work for me. I tried several but 10 was the only one I could get to install

cprezzi, 2019-04-23:
I've updated the wiki. Thanks for suggesting!

closed 2019-04-23 by cprezzi

pen25, 2019-04-23:
Awesome thx.