LaserWeb/lw.comm-server #75: rpi buster not working

bokky, 2019-07-21:
I wanted to move away from my fixed linux pc to raspberrypi setup but I can not get it to find my USB connected laser. The reason for that is it gives tons of errors compiling serialport.

I followed the node10 manual here on github, which is written for debian wheezy. I noticed that Rpi updated their noobs and raspbian images to debian buster in June, most probably that has to do with my errors. 

Anybody successfully build it on buster?

Here are my specs:
Rpi v3
node -v : v10.16.0
npm -v : 6.9.0

Errorlog when running "npm install"  (it is long, I know. Just want to be complete)

root@laserpi:~/lw.comm-server# npm install

> serialport@6.2.2 install /root/lw.comm-server/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.0 runtime=node arch=arm platform=linux)
make: Entering directory '/root/lw.comm-server/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2208:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:41:49: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated: Use Isolate version [-Wdeprecated-declarations]
   v8::String::Utf8Value path(info[0]->ToString());
                                                 ^
In file included from /root/.node-gyp/10.16.0/include/node/v8.h:26,
                 from /root/.node-gyp/10.16.0/include/node/node.h:63,
                 from ../../nan/nan.h:52,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/root/.node-gyp/10.16.0/include/node/v8.h:2892:28: note: declared here
                   explicit Utf8Value(Local<v8::Value> obj));
                            ^~~~~~~~~
/root/.node-gyp/10.16.0/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/serialport.cpp:48:53: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   v8::Local<v8::Object> options = info[1]->ToObject();
                                                     ^
In file included from /root/.node-gyp/10.16.0/include/node/node.h:63,
                 from ../../nan/nan.h:52,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/root/.node-gyp/10.16.0/include/node/v8.h:10046:15: note: declared here
 Local<Object> Value::ToObject() const {
               ^~~~~
../src/serialport.cpp:78:69: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
                                                                     ^~~~~~~~~~~~~
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:95:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
In file included from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../nan/nan.h:1655:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:113:53: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   v8::Local<v8::Object> options = info[1]->ToObject();
                                                     ^
In file included from /root/.node-gyp/10.16.0/include/node/node.h:63,
                 from ../../nan/nan.h:52,
[... 188 lines cut ...]
/root/.node-gyp/10.16.0/include/node/v8.h:9502:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfo<node::ObjectWrap>::Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<node::ObjectWrap>&)’} to ‘Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<void>&)’} [-Wcast-function-type]
                reinterpret_cast<Callback>(callback), type);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/.node-gyp/10.16.0/include/node/v8.h: In instantiation of ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = Nan::ObjectWrap; T = v8::Object; typename v8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)]’:
../../nan/nan_object_wrap.h:66:61:   required from here
/root/.node-gyp/10.16.0/include/node/v8.h:9502:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfo<Nan::ObjectWrap>::Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)’} to ‘Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<void>&)’} [-Wcast-function-type]
  CXX(target) Release/obj.target/serialport/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/root/lw.comm-server/node_modules/serialport/build'

> electron@1.8.8 postinstall /root/lw.comm-server/node_modules/electron
> node install.js

Downloading SHASUMS256.txt
[============================================>] 100.0% of 5.74 kB (5.74 kB/s)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: 7zip-bin-win@2.1.0 (node_modules/7zip-bin-win):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for 7zip-bin-win@2.1.0: wanted {"os":"win32","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: 7zip-bin-mac@1.0.1 (node_modules/7zip-bin-mac):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for 7zip-bin-mac@1.0.1: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 630 packages from 320 contributors and audited 1926 packages in 160.077s
found 78 vulnerabilities (34 low, 25 moderate, 18 high, 1 critical)
  run `npm audit fix` to fix them, or `npm audit` for details
root@laserpi:~/lw.comm-server#

cprezzi, 2019-07-22:
The tutorial sais Raspbian Jessie, but I think it also works with Stretch. Never tried Buster.
Why don't u just use a Jessie?

bokky, 2019-07-23:
i used a Rpi3 B+ but Jessie is not running on it, you need a newer distro, therefor I took the lastest.

Digging in my shelve I found a Rpi3 B, jessie boots fine but after following the lwcomm manual it is still failing on serialport module, now with different errors (see at the end).

I used the last jessie-lite image (July-2017), which give me:
node -v : v10.16.0
npm -v : 6.9.0

I also build the docker container but at compile time it gives me the same error messages at the serial module.
So can anybody tell my what magical combo you need to have of OS, node and npm and serialport to make it work?

pi@laserpi:~/lw.comm-server $ npm install

> serialport@6.2.2 install /home/pi/lw.comm-server/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.0 runtime=node arch=arm platform=linux)
make: Entering directory '/home/pi/lw.comm-server/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:41:49: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /home/pi/.node-gyp/10.16.0/include/node/v8.h:2891): Use Isolate version [-Wdeprecated-declarations]
   v8::String::Utf8Value path(info[0]->ToString());
                                                 ^
../src/serialport.cpp:48:53: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/10.16.0/include/node/v8.h:10046): Use maybe version [-Wdeprecated-declarations]
   v8::Local<v8::Object> options = info[1]->ToObject();
                                                     ^
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:95:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1655) [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:113:53: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/10.16.0/include/node/v8.h:10046): Use maybe version [-Wdeprecated-declarations]
   v8::Local<v8::Object> options = info[1]->ToObject();
                                                     ^
../src/serialport.cpp: In function ‘void EIO_AfterUpdate(uv_work_t*)’:
../src/serialport.cpp:150:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1655) [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
../src/serialport.cpp: In function ‘void EIO_AfterClose(uv_work_t*)’:
../src/serialport.cpp:188:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1655) [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
../src/serialport.cpp: In function ‘void EIO_AfterFlush(uv_work_t*)’:
../src/serialport.cpp:231:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1655) [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:250:53: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated (declared at /home/pi/.node-gyp/10.16.0/include/node/v8.h:10046): Use maybe version [-Wdeprecated-declarations]
   v8::Local<v8::Object> options = info[1]->ToObject();
                                                     ^
../src/serialport.cpp: In function ‘void EIO_AfterSet(uv_work_t*)’:
../src/serialport.cpp:285:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1655) [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
../src/serialport.cpp: In function ‘void EIO_AfterGet(uv_work_t*)’:
../src/serialport.cpp:336:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1655) [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
../src/serialport.cpp: In function ‘void EIO_AfterGetBaudRate(uv_work_t*)’:
../src/serialport.cpp:383:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1655) [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
../src/serialport.cpp: In function ‘void EIO_AfterDrain(uv_work_t*)’:
../src/serialport.cpp:424:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1655) [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/poller.o
../src/poller.cpp: In static member function ‘static void Poller::onData(uv_poll_t*, int, int)’:
[... cut ...]
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for 7zip-bin-win@2.1.0: wanted {"os":"win32","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: 7zip-bin-mac@1.0.1 (node_modules/7zip-bin-mac):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for 7zip-bin-mac@1.0.1: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 630 packages from 320 contributors and audited 1926 packages in 158.419s
found 78 vulnerabilities (34 low, 25 moderate, 18 high, 1 critical)
  run `npm audit fix` to fix them, or `npm audit` for details

cprezzi, 2019-07-24:
I cannot see any errors, only warnings. It should work like this.

bokky, 2019-07-24:
The service start fine and does not crash, the site loads fine but when I look in Comms -> machine connection there is nothing to select from in the drop down menu. 
Do I need to hard code it somewhere which my usb connected port is (/dev/ttyUSB0) ?
Or do I need to set something else?

On my desktop full laserweb works nicely and porst are immediately available.

bokky, 2019-07-24:
Ok seems that I have found it... but weired...

Exported my desktop config of laserweb, imported it into rpi laserweb. Then I got a usb port avaiable (not the correct one)
Then I did server connection -> to my pi port 8000 -> connect
Now I have the correct ports available in USB, I managed to get a successful connect 

Stil encountered out of bound errors but I should be able to solve that myself, no time today to figure that out.

closed 2019-07-28 by bokky