LaserWeb/lw.comm-server #76: Not Installable on Raspberry Pi

joe9439, 2019-10-10:
Not able to be installed using the wiki or website instructions on raspberry pi 2 or pi 3b+.

I've tried all kinds of things. I've tried many different versions of raspbian on both raspberry pis. It just doesn't work.

cprezzi, 2019-10-10:
Can you post the error messages?

Benjin454, 2019-10-14:
Not an expert. But it looks like it may have to do with the node and npm versions.    
[terminal log.txt](https://github.com/LaserWeb/lw.comm-server/files/3725339/terminal.log.txt)

Benjin454, 2019-10-14:
Tried with the "correct" npm and node versions using NVM with no luck. Logs Attached.
[terminal log.txt](https://github.com/LaserWeb/lw.comm-server/files/3725958/terminal.log.txt)

[2019-10-14T17_55_06_851Z-debug.log](https://github.com/LaserWeb/lw.comm-server/files/3725959/2019-10-14T17_55_06_851Z-debug.log)
[2019-10-14T17_55_08_414Z-debug.log](https://github.com/LaserWeb/lw.comm-server/files/3725961/2019-10-14T17_55_08_414Z-debug.log)
[2019-10-14T17_56_26_060Z-debug.log](https://github.com/LaserWeb/lw.comm-server/files/3725962/2019-10-14T17_56_26_060Z-debug.log)
[2019-10-14T17_56_27_656Z-debug.log](https://github.com/LaserWeb/lw.comm-server/files/3725963/2019-10-14T17_56_27_656Z-debug.log)

cprezzi, 2019-10-15:
Your log sais that you are using Node 10.15.1 (not 10.16.3). Change the node version in package.json to 10.15.1 and try again.

cprezzi, 2019-10-15:
If that doesn't help, then you need to update to node 10.16.3.

Benjin454, 2019-10-15:
It doesn't work when removing that line. But it appears to be working after using the package.json you just edited! Thanks for all your hard work on this project! Great software

closed 2019-10-15 by cprezzi