LaserWeb/lw.comm-server #77: runing lw.comm-server on OrangePi Zero websocket problem

abrakadobr, 2019-12-24:
so, i have 3 axis mini cnc with GRBL, i was success to run it with my laptop (ubuntu) using LaserWeb4 appimage. now i want to move LW4 to OrangePi Zero Armbian Bionic with Linux 5.3.9-sunxi.
i got some headache with installing, but finally i got all runing, even small "serialtest.js" shows me /dev/ttyUSB0 from cnc.
i'm using `nvm` to mange node versions, `npm i` was success only with node v10, and not under root (only under user). after with `node server.js` all runs, and i'm getting frontend on http://192.168.1.90:8000 (board load ip) from my laptop. 
But frontend can't connect to backend with websocket.
I'm ok with js/nodejs, and inserting few "console.log" to "connection points" i see no "frontend app connection".
also chrome/firefox network debuggers shows that there is no websocket connection.
i'll keep go deeper, but any help is welcome =)

abrakadobr, 2019-12-24:
ok... as usual, after 3 days of researching, i just changes "server name" from "localhost:8000" to "192.168.1.90:8000" (local ip of board) and all start working =))))))))))
thanks, LaserWeb team, for really nice software!

closed 2019-12-24 by abrakadobr

cprezzi, 2019-12-24:
This is documented in the wiki.
see https://github.com/LaserWeb/lw.comm-server/wiki/Manual-Installation-(RasPi)#install--start-the-frontend