LaserWeb/lw.comm-server #82: FEATURE: Clean up the horrible build system

janschiefer, 2020-04-22:
TODO: Merge lw.comm-server, lw.comm-server#electron_bundler and LaserWeb4 into one repsitory and remove duplicates.

TODO: Move from npm to yarn.

TODO: Migrate to newer package versions of dependencies (especially electron and electron-builder).

TODO: Implement .deb building for Linux.

cprezzi, 2020-04-23:
There are reasons why it was setup like this. 
- lw.comm-server was made to be an independant websocket to machine gateway that could serve different frontends.
- The master branch was only for development.
- The electron_builder branch was for building different OS versions of LW4.
- The Windows standalone EXE makes >90% of usage.

Sure, it would be nice to migrate to actual dependencies, but this needs a bigger effort.

Isn't .deb only for Debian but AppImage for all sorts of Linux?

joesantosio, 2022-09-25:
My opinion as a newcomer to the project (just my 2 cents, take it only as an opinion):
- I like the separation of concerns of having gui separated from the api, in matter of fact, I think it should even go further and fully separate gui from api. API shouldn't know about gui. Gui should connect to the server by host and that is it
- I don't find any point in moving to yarn
- Definitely dependencies are quite outdated and even moving the app gui to Tauri could be a very interesting move
- You could use docker cross app (I've made a PR to make it work)

What about moving the API to something more resilient as golang? Is that of any interest to you? It would provide a better build system, a better security (code wise), a single file build without dependencies to run cross platform, less dependency hell... I wouldn't mind to work on that new version and research on how we could do it slowly.

.deb is used also by ubuntu systems I think

malcolmmp, 2024-06-11:
I know I'm replying to a very old issue here, but I'd like to share the work on the build system that I've recently done, for anyone who may be interested:

https://github.com/malcolmmp/LaserWeb4-Binaries/tree/electron-builder

I also made a post about it on the makerforums website here:

https://forum.makerforums.info/t/building-lw4-binaries-in-2024/90044

The work is by no means complete or ready for mainline, but it does build cleanly on Lin/Win/Mac with Node 22.

I essentially pulled in the latest versions of everything, added some of the work done for edmweb, and then ported and updated the electron_builder specific stuff over to LaserWeb4-Binaries.

If it's of interest, I'd love to clean up the messy bits and submit some pull requests!