LaserWeb/lw.comm-server #85: Remove Laserweb from Raspberry

Dane809, 2020-08-24:
Can´t remove laserweb from raspberry pi. How can I do that?

cprezzi, 2020-08-24:
Did you try to delete the lw.comm-servcer folder?

Dane809, 2020-08-24:
Yes. The version is still 4.0.136
Want to install the newest version, because my settings -> tools don´t work correctly. Can´t open and save the machine settings (other issue)

Thx for your answer @cprezzi 
Great support!

cprezzi, 2020-08-24:
4.0.136 is the actual version of lw.comm-server. Nothing newer available.

Where do you run the frontend? On the Raspberry Pi or another PC in the network?

Dane809, 2020-08-24:
Another PC in the network. 
Tools and macros are not "clickable" from the PC

cprezzi, 2020-08-24:
Then you need to install / reinstall the LaserWeb binary (EXE) on that network PC from here:
https://github.com/LaserWeb/LaserWeb4-Binaries/releases

Dane809, 2020-08-25:
Sounds like there is a bug in the webinterface of the raspberry?
Today I use the ui with chrome on my pc with the raspberry ip-adress and port 8000.

cprezzi, 2020-08-25:
I have now updated the hosted frontend files in the app folder of lw.comm-server. Try to make a "git pull" on your raspi (or download and unzip).

Dane809, 2020-08-27:
Hey, 
will try it today. 
Thx a lot!

closed 2020-08-27 by Dane809