LaserWeb/lw.comm-server #87: Job Abort on Marlin sends M112, this halts the controller and forces a manual reset by the user.

descipher, 2020-11-23:
It would be better to send nothing and allow a resume by the user. A job abort is restarted on the client end and if the user does resend it starts from the beginning of the job. 

Recommend removing the following from server.js:

-                machineSend('M112\n'); // abort
-                writeLog('Sent: M112', 2);

and have the following action:

            case 'marlin':
                paused = true;
                break;

cprezzi, 2020-11-23:
Job Abort is meant to be a emergency stop (immediate, in the middle of a move), in case of a serious problem.
What you describe is the the Job Pause function, which can be resumed afterwards.

descipher, 2020-11-23:
Hi Claudio,

 

I understand the functions, however if this were grbl you can simply send $X and it’s communication is restored. 

On marlin the CPU is halted, there is no communication and it’s dead, you need a power cycle or reset.

I can see more graceful ways of handling the abort on Marlin. 

For example we can send a shut off the laser unlock the motors and still communicate with it from there.

closed 2021-01-30 by cprezzi