LaserWeb/lw.comm-server #91: socket.io - Unsupported protocol version

filka96, 2021-03-16:
Using https://laserweb.yurl.ch/documentation/installation/36-install-raspberry-pi
network errors
request:
```
GET | http://192.168.100.214:8000/socket.io/?EIO=3&transport=polling&t=NWyFsWL
```
response:
```
400 - Bad Request
{
  "code": 5,
  "message": "Unsupported protocol version"
}
```

filka96, 2021-03-16:
fix from https://socket.io/docs/v3/troubleshooting-connection-issues/:
```
var io = websockets(app, {
  allowEIO3: true // false by default
});
```

cprezzi, 2021-09-27:
This issue probably comes from installing a newer version of node.js (>10), which uses a newer version of socket.io.
It would make sense to migrate the code to the actual node,js and dependencies, but that takes a lot of work, which I don't have time for.

closed 2021-09-27 by filka96