LaserWeb/lw.comm-server #92: How can I build a Windows standalone exe ?

mfadel85, 2021-04-02:
Hello everyone,
How can I build a Windows standalone binary EXE, to run the project without the need to any browser?
I didn't find any document or guidance

Thank you very much

mfadel85, 2021-04-02:
is npm run dist enough?
I tried it, it didn't work  it says ncp is not defined I am working on this.

cprezzi, 2021-04-06:
You need to use the electron_bundler branch to compile an executable. Check out the scripts in package.json. 
If you don't have "ncp" in your shell, you can change the script to "cp" (which is available on most shells).

mfadel85, 2021-04-14:
@cprezzi Thank you very much

mfadel85, 2021-04-14:
I have a problem in the last step,  when running npm run dist
Error: Windows icon size must be at least 256x256, please fix "...\lw.comm-server\app\favicon.ico"
but the icon is being generated automatically, how can I change its setting to be 256x256??
I didn't find it any place to change the resolution of the favicon.

cprezzi, 2021-04-14:
There is a wrong icon defined for windows in package.json line 74 (see: https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/package.json#L74).
Try to change this line from:
      "icon": "app/favicon.ico"
to
      "icon": "build/icon.ico"

mfadel85, 2021-04-15:
@cprezzi thank you, it worked

closed 2021-04-15 by mfadel85