LaserWeb/lw.comm-server #95: LaserWeb causes a lot of udevadm instances to start up

PetteriAimonen, 2021-09-26:
I've noticed that after running for a long time, lw.comm-server starts clogging up the server with `udevadm` instances.
These are started by `node_modules/serialport/lib/bindings/linux-list.js` to scan for serial ports, but normally the `udevadm info -e` command takes just 100 ms and shouldn't be much of a problem.

It appears that the main code of `lw.comm-server` starts a loop that checks for serial ports every 2 seconds on every connection:
https://github.com/LaserWeb/lw.comm-server/blob/master/server.js#L166

But that loop appears to be never stopped - i.e. after 100 connections, lw.comm-server rescans for serial ports 50 times per second. I'm not very familiar with javascript though, so my analysis may be incorrect.

cprezzi, 2021-09-27:
Nice find! 
Every Connect starts it's own interval, instead of reusing the existing interval (or stopping and restarting it).
So either we should stop the interval before (re)creating it or only create it, if it is not already created.

I will add that to the next version.

cprezzi, 2021-09-27:
Done. Would be nice it you could test.

PetteriAimonen, 2021-09-27:
Thanks! Looks good from code but I'll test next time I'm at the hackerspace.

PetteriAimonen, 2021-09-29:
I can confirm that this bug is now fixed and that detection of new serial ports still works also (at the 2 second interval as it should).

closed 2021-09-29 by PetteriAimonen