LaserWeb/lw.comm-server #96: Commit f43136a introduced bugs into app/index.js

cptskippy, 2021-09-30:
There are multiple issues starting on line 44675.  I'm not familiar with React but these errors seem to occur when referenced resources cannot be loaded?

https://github.com/LaserWeb/lw.comm-server/commit/f43136a47ac892008e7e5e44ef22355d84dd4c64

cprezzi, 2021-10-01:
Line 44675 of which file?

cptskippy, 2021-10-01:
The issue is with the [/app/index.js](https://github.com/LaserWeb/lw.comm-server/blob/4c02afff416531ab82a0a03da1e9d27961e05939/app/index.js) file on multiple lines starting on line 44675:

	Line 44675: <<<<<<< HEAD
	Line 44716: <<<<<<< HEAD
	Line 44952: <<<<<<< HEAD
	Line 46113: <<<<<<< HEAD
	Line 46733: <<<<<<< HEAD
	Line 68457: <<<<<<< HEAD
	Line 68540: <<<<<<< HEAD
	Line 68639: <<<<<<< HEAD
	Line 68709: <<<<<<< HEAD
	Line 69076: <<<<<<< HEAD
	Line 69684: <<<<<<< HEAD
	Line 69959: <<<<<<< HEAD
	Line 70191: <<<<<<< HEAD
	Line 70824: <<<<<<< HEAD
	Line 72331: <<<<<<< HEAD
	Line 73217: <<<<<<< HEAD
	Line 74777: <<<<<<< HEAD
	Line 75077: <<<<<<< HEAD
	Line 75262: <<<<<<< HEAD
	Line 75401: <<<<<<< HEAD
	Line 75556: <<<<<<< HEAD
	Line 76200: <<<<<<< HEAD
	Line 76483: <<<<<<< HEAD
	Line 76666: <<<<<<< HEAD
	Line 76823: <<<<<<< HEAD
	Line 77822: <<<<<<< HEAD
	Line 78319: <<<<<<< HEAD
	Line 78968: <<<<<<< HEAD
	Line 79143: <<<<<<< HEAD
	Line 206564: <<<<<<< HEAD
	Line 206587: <<<<<<< HEAD
	Line 206600: <<<<<<< HEAD
	Line 206611: <<<<<<< HEAD
	Line 206626: <<<<<<< HEAD

cprezzi, 2021-10-01:
Ok, thanks. I will check that.
Probably it still works on my system because of some cached files.

closed 2021-11-30 by cprezzi