LaserWeb/lw.comm-server #101: No server / machine connection running on Raspberry Pi 3B+

t-richter2, 2022-02-05:
I'm trying to get this to run on a 3B+
I followed the installation guide [here](https://github.com/LaserWeb/lw.comm-server/wiki/Manual-Installation-(RasPi)) line by line.
I also tested this with a fresh buster and bullseye image.
The result is somewhat strange?

On the Pi I can start the server and get this output:

[image]

I can connect to the fontend via browser, however I can't add any machine.

[image]

Note I never get the green "connected to server" message as mentioned in the wiki

[image]

This is also the case when trying to connect from the LaserWeb4 app running on macOS

[image]

easytarget, 2022-02-05:
Thankyou. good report :-), 

What I immediately see is that you are not putting the correct SERVER IP address in the 'Connect' tab:

You should use the same URL as you type in the browsers address bar, eg the 'http://192.168.0.222:8000' that you see in the comm-server output.

If you still cant get that working once you have the correct IP set there..  I need more info:

>    Versions
 Frontend: xxxx
 Backend: (should say 'not connected')
 Latest release: xxxx

This is could be a version missmatch, we changed LW late last year to use an updated socket.io comms protocol, and this means the frontend and backend need to be in sync.. but if you followed the instructions then you should be ok..

t-richter2, 2022-02-05:
Changing the IP did not change the behavior.

[image]

Version mismatch could be the problem.
Server says: LaserWeb Comm Server 4.1.000
Browser says: Laserweb 4.0.999

My git is up-to-date with the master branch

Don't know how to get the backend status, though, sorry

FrenkK, 2022-02-05:
Same problem here. One tidbit of info that might help: If I connect in the web browser using the ip given in console, it connects, but not from the pc app. I can't get that to connect at all.

FrenkK, 2022-02-05:
Dug some more in the latest app (.999_138).
I turned on the dev tools and get the following error message that repeats:
"Failed to load http://laserweb:8000/socket.io/?EIO=3&transport=polling&t=NxAf7hY: The value of the 'Access-Control-Allow-Origin' header in the response must not be the wildcard '*' when the request's credentials mode is 'include'. Origin 'http://127.0.0.1:8000' is therefore not allowed access. The credentials mode of requests initiated by the XMLHttpRequest is controlled by the withCredentials attribute."

easytarget, 2022-02-05:
The version missmatch you see is actually OK. lw.comm-server got a 4.1 upgrade before we have had a chance to (fully) get 4.1 frontend out of the door. 

the console messages you see @FrenkK look more relevant. I'll investigate

@t-richter2 ; can you do the same, rt-click and open developer tools, click the console tab and see if any relevent messages pop up there, you might see some react.js errors too, ignore them.

t-richter2, 2022-02-05:
I doubt this is related, but I turned on "Enable Developer Debug", after that I had success, at least  partially?
[image]

I feel a bit stupid now, but I haven't touched anything else?

t-richter2, 2022-02-05:
Sorry, even more stupid now.
It must have been the IP settings, that fixed it for me in the first place.
Might have forgotten to properly restart the server after last test :cough:
"No supported firmware" is another topic, i guess.
Many thanks for your support.

easytarget, 2022-02-05:
> "No supported firmware" is another topic, i guess.

Yep; You got it :grinning: