LaserWeb/lw.comm-server #102: Working with reverse proxy

fooker, 2022-02-25:
I tried to use the laserweb server behind a reverse proxy (`nginx` in this case) but it looks like the internal IP is leaked somewhere.
Even if the app is accessed via `laserweb.home.example.com:80` there are requests from the browser to the original binding port `8080`.

It looks like the app does not honor the according headers.

cprezzi, 2022-02-26:
The server was not built nor tested to be used from outside of the network. There are two communication channels used. On is serving the frontend via the nodejs webserver on the port that is defined in config.js. The second is a websocket connection that is used from the frontend to the backend.

Which of those communications was not working? 
- Was the frontend website deliverd to the client? 
- Was the server setting on client side set correctly?
- Was the connection from the client detected by the server (console output)?

fooker, 2022-02-27:
@cprezzi Thanks for the quick answer.

My intend is not to use this outside a network. I'm starting the server as an unprivileged user, The server is bound to port 8080 on 127.0.0.1 only. The reverse proxy is running on the same machine and provides the service on port 80. I'm doing this because the  same host serves other web content already on a different domain name.

> * Was the frontend website deliverd to the client?

Yes, the forntend is delivered through the reverse proxy without any problems.

> * Was the server setting on client side set correctly?

I was not aware of the existence of this setting. It is set to `localhost:8000` by default. Changing this to the same address used to open the web-app solves the problem.

> * Was the connection from the client detected by the server (console output)?

There is no output per connection.

---

Maybe this could be solved by altering the frontend to use the address used to serve the frontend as a default for the server setting.

cprezzi, 2022-02-27:
It has to be changeable in the frontend for the case when LaserWeb is installed on the client (mostly Windows EXE), but needs to connect to a server (like Raspberry Pi).

closed 2022-02-27 by cprezzi

fooker, 2022-03-01:
I know that it needs to be changeable. But the default is wrong and may be changed to the correct value.

cprezzi, 2022-03-02:
Why do you think that the default is wrong? 
It is correct for the main usecase, which is the Windows EXE with self contained server.

fooker, 2022-03-02:
> It is made for the main usecase, which is the Windows EXE with self contained server. Having the server on a different machine is a very rare case at our userbase. But sure, gettig the IP from the "server" that delivers the frontend would work on most cases. I'll add this to the "nice to have" list ;)

My thought was to use something like `window.location.host` but I have no clue about the electron environment.