LaserWeb/lw.comm-server #105: Laser test crashes server

helifino, 2022-04-29:
***************************************************************
        ---- LaserWeb Comm Server 4.1.000 ----
***************************************************************
  Use  http://192.168.2.105:8000 to connect to this server.
[...]
***************************************************************

Server binding to all local IP addresses on port: 8000
App connected! (id=0, ip=192.168.2.6)
Connect (0) Sending Interfaces list: USB,ESP8266,Telnet
INFO: Requesting Server Config
Connect(0) Sending Ports list: /dev/ttyUSB0,/dev/ttyS0,/dev/ttyS1,/dev/ttyS10,/dev/ttyS11,/dev/ttyS12,/dev/ttyS13,/dev/ttyS14,/dev/ttyS15,/dev/ttyS16,/dev/ttyS17,/dev/ttyS18,/dev/ttyS19,/dev/ttyS2,/dev/ttyS20,/dev/ttyS21,/dev/ttyS22,/dev/ttyS23,/dev/ttyS24,/dev/ttyS25,/dev/ttyS26,/dev/ttyS27,/dev/ttyS28,/dev/ttyS29,/dev/ttyS3,/dev/ttyS30,/dev/ttyS31,/dev/ttyS4,/dev/ttyS5,/dev/ttyS6,/dev/ttyS7,/dev/ttyS8,/dev/ttyS9
INFO: Connecting to USB,/dev/ttyUSB0,115200
Sent: ctrl-x
INFO: Connected to /dev/ttyUSB0 at 115200
GRBL detected (1.1f)
laserTest: Power 1, Duration 0, maxS 1
/home/alex/lw.comm-server/server.js:3013
                        appSocket.emit('laserTest', power);
                        ^

ReferenceError: appSocket is not defined
    at laserTest (/home/alex/lw.comm-server/server.js:3013:25)
    at Socket.<anonymous> (/home/alex/lw.comm-server/server.js:2015:9)
    at Socket.emit (events.js:314:20)
    at Socket.emitUntyped (/home/alex/lw.comm-server/node_modules/socket.io/dist/typed-events.js:69:22)
    at /home/alex/lw.comm-server/node_modules/socket.io/dist/socket.js:466:39
    at processTicksAndRejections (internal/process/task_queues.js:79:11)

cprezzi, 2022-04-30:
Thats a bug on line 3013 of server.js.
Instead of "appSocket.emit..." it should be "io.sockets.emit...".
I will correct that in the next version.

helifino, 2022-04-30:
Side note, the version number on the web interface is mismatched (displays 4.0.999)

helifino, 2022-04-30:
laserTest: Power off
/home/alex/lw.comm-server/server.js:3141
                appSocket.emit('laserTest', 0);
                ^

ReferenceError: appSocket is not defined
    at laserTest (/home/alex/lw.comm-server/server.js:3141:17)

helifino, 2022-04-30:
Changes made in local copy, laser still doesn't fire and no status message in web interface

cprezzi, 2022-05-01:
It is correct, that the frontend version doesn't macht the server version.
After connecting the server from the frontend, both versions will be shown.

All the appSocket.emit in the LaserTest function need to be changes to io.sockets.emit.

cprezzi, 2022-05-01:
What firmware do you use?
On grbl, the default max S value is 1000, so you would need to set the max S value to 1000. 

Do you use a Diode Laser or CO2?

cprezzi, 2022-05-01:
It is correct, that "check size" does not activate the laser. 
This is for security reason. On a CO2 laser you would not want to activate the laser on "check size".

On a diode laser, you can just set the laser test duration to 0 and press "Laser Test" before "check size" and again after it.

helifino, 2022-05-01:
This should be a per-machine configuration option. Like, a checkbox next to "Check-Size Power".

cprezzi, 2022-05-02:
I'll add this to the wish list...

closed 2022-05-02 by cprezzi