LaserWeb/lw.comm-server #106: Feature request - linuxen

helifino, 2022-05-15:
Remove user/directory hardcoding in *.service and build up using a 'preinstall' script, parsing $USER and $CWD environment variables instead. Will be more portable, for any debian derivative which uses the .service file, not just pi.

cprezzi, 2022-05-17:
I'm not a linux epert, but would be happy to include your chanes. 
Please make a pull requenst or send me the needed changes.

joesantosio, 2022-09-25:
What about relying more on docker? I like systemd but at the same time it means that you need to handle the dependencies by yourself be it node or the project dependencies or even environment variables. Docker makes it easy to do something like building from dockerhub and `--restart=always` to run it on boot and restart if any issue.

cprezzi, 2022-09-27:
Our main userbase are Windows Users. They prefer a simple installer/exe and don't want to deal with docker.
But Docker might be a good solution for installation of lw.comm-server on Raspi & Linux.